nordicsemi/Android-nRF-Connect-Device-Manager #16: Cannot disconnect from BLE device

lambo81, 2021-09-23:
From what I can tell, ever since I added the device manager to my gradle, I am not able to disconnect from my device, despite getting the disconnect callbacks.  The part of the log that says:  "bta_gattc_mark_bg_conn unable to find the bg connection mask for: d4:3f:9b:09:4e:5b" might be of interest.

Right now I am just disconnecting under normal conditions (not DFU related).  The device will only disconnect if I disable Bluetooth on my Android device.

Here is my gradle:

    // https://mvnrepository.com/artifact/no.nordicsemi.android/ble-livedata
    implementation 'no.nordicsemi.android:ble-livedata:2.3.1'
    // https://mvnrepository.com/artifact/no.nordicsemi.android.support.v18/scanner
    runtimeOnly 'no.nordicsemi.android.support.v18:scanner:1.5.0'

    // https://mvnrepository.com/artifact/no.nordicsemi.android/mcumgr-core
    implementation 'no.nordicsemi.android:mcumgr-ble:0.13.0-beta04'

Here is the log:

2021-09-23 13:11:18.517 15883-15883/com.r2f.android.orthofit.debug D/BluetoothGatt: cancelOpen() - device: D4:3F:9B:09:4E:5B
2021-09-23 13:11:18.518 14579-14653/com.android.bluetooth D/BtGatt.GattService: clientDisconnect() - address=D4:3F:9B:09:4E:5B, connId=8
2021-09-23 13:11:18.518 14579-14616/com.android.bluetooth E/bt_btif: bta_gattc_mark_bg_conn unable to find the bg connection mask for: d4:3f:9b:09:4e:5b
2021-09-23 13:11:18.519 14579-14600/com.android.bluetooth D/BtGatt.GattService: onDisconnected() - clientIf=8, connId=8, address=D4:3F:9B:09:4E:5B
2021-09-23 13:11:18.519 15883-15939/com.r2f.android.orthofit.debug D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=D4:3F:9B:09:4E:5B
2021-09-23 13:11:18.521 15883-15939/com.r2f.android.orthofit.debug D/BluetoothGatt: close()
2021-09-23 13:11:18.521 15883-15939/com.r2f.android.orthofit.debug D/BluetoothGatt: unregisterApp() - mClientIf=8
2021-09-23 13:11:18.526 14579-14653/com.android.bluetooth D/BtGatt.GattService: unregisterClient() - clientIf=8
2021-09-23 13:11:18.617 15883-15883/com.r2f.android.orthofit.debug D/BleRepository: ConnectionObserver onDeviceDisconnecting() com.r2f.android.repository.BikeControl@d051f71 D4:3F:9B:09:4E:5B
2021-09-23 13:11:18.621 15883-15883/com.r2f.android.orthofit.debug D/BleRepository: ConnectionObserver onDisconnected() com.r2f.android.repository.BikeControl@d051f71 D4:3F:9B:09:4E:5B 0

lambo81, 2021-09-24:
I will write a new issue that is more clear on how to reproduce

closed 2021-09-24 by lambo81

philips77, 2021-09-24:
FYI: I've released 0.13.0-beta06 yesterday, which depends on BLE lib 2.3.1.

As to your issue, remember, that the physical connection is maintained by Android, and each client (`BluetoothGatt` or `BleManager`) has a virtual connection. The physical connection is terminated a second after the last client has disconnected. You may be connected to the same device from multiple apps, or even multiple clients in the same app (e.g. main connection and the client built-in the `McuMgrBleTransport`. To make sure the device disconnects, you may sent Reset command using `DefaultManager`, which will cause the device to reset, and therefore disconnect. You may also create a custom command just to disconnect, without resetting, similar way to how erase storage command has been added in Zephyr ([here](https://github.com/zephyrproject-rtos/zephyr/pull/36548), but use user space for groups (PERUSER)).

lambo81, 2021-09-24:
Thanks for the tips.