nordicsemi/Android-nRF-Connect-Device-Manager #19: Mcu Mgr Error: NO_MEMORY (2)

ayecarranda, 2021-11-09:
I have been trying to upgrade my device through my Android app using this library, and I continually get the following error in the UPLOAD stage: 
`2021-11-09 15:07:49.439 6269-6269/com.somewearlabs.sw.debug D/[main] FirmwareUpdater: upgrade failed=UPLOAD; error=io.runtime.mcumgr.exception.McuMgrErrorException: Mcu Mgr Error: NO_MEMORY (2)`. 

I have tried using the standalone NRF Connect Device Manager app, and using Advanced > Select File > Upload, and I get the same NO_MEMORY error there too. I tried restarted the device I'm trying to upgrade, and also tried repeating the upload with fresh images each time. I still get this error.

Could someone please help me with a workaround?

Version: 0.13.0-beta-08
Android Version: 10

More logs attached: 

```
2021-11-09 15:06:47.160 6269-6269/com.somewearlabs.sw.debug D/[main] onStateChanged: old=VALIDATE, new=RESET
2021-11-09 15:06:47.161 6269-6269/com.somewearlabs.sw.debug D/[main] BleUtility: startFirmwareUpdateTimer: start. Expect DFU related disconnect...;
2021-11-09 15:06:47.163 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Sending (10 bytes) Header (Op: 2, Flags: 0, Len: 2, Group: 0, Seq: 1, Command: 5) CBOR {}
2021-11-09 15:06:47.163 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48)
2021-11-09 15:06:47.165 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 02-00-00-02-00-00-01-05-BF-FF
2021-11-09 15:06:47.190 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-02-00-00-01-05-BF-FF
2021-11-09 15:06:47.194 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Received Header (Op: 3, Flags: 0, Len: 2, Group: 0, Seq: 1, Command: 5) CBOR {}
2021-11-09 15:06:47.936 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=5 device=C4:EE:1A:38:C8:50
2021-11-09 15:06:47.942 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: [Callback] Connection state changed with status: 8 and new state: 0 (DISCONNECTED)
2021-11-09 15:06:47.943 6269-6269/com.somewearlabs.sw.debug W/[main] McuMgrBleTransport: Error: (0x8): GATT CONN TIMEOUT
2021-11-09 15:06:47.943 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Disconnected
2021-11-09 15:06:47.969 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.close()
2021-11-09 15:06:47.970 6269-6269/com.somewearlabs.sw.debug D/BluetoothGatt: close()
2021-11-09 15:06:47.975 6269-6269/com.somewearlabs.sw.debug D/BluetoothGatt: unregisterApp() - mClientIf=5
2021-11-09 15:06:47.982 6269-6269/com.somewearlabs.sw.debug D/[main] onDisconnected - could be because of RESET stage
2021-11-09 15:06:47.982 6269-6269/com.somewearlabs.sw.debug I/[main] Reset: Device disconnected
2021-11-09 15:07:47.247 6269-6269/com.somewearlabs.sw.debug D/[main] raonStateChanged: old=RESET, new=UPLOAD
2021-11-09 15:07:47.275 6269-15701/com.somewearlabs.sw.debug D/[pool-29-thread-1] McuMgrBleTransport: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2021-11-09 15:07:47.276 6269-15701/com.somewearlabs.sw.debug D/BluetoothAdapter: STATE_ON
2021-11-09 15:07:47.279 6269-15701/com.somewearlabs.sw.debug D/BluetoothGatt: connect() - device: C4:EE:1A:38:C8:50, auto: false
2021-11-09 15:07:47.279 6269-15701/com.somewearlabs.sw.debug D/BluetoothAdapter: isSecureModeEnabled
2021-11-09 15:07:47.279 6269-15701/com.somewearlabs.sw.debug D/BluetoothGatt: registerApp()
2021-11-09 15:07:47.280 6269-15701/com.somewearlabs.sw.debug D/BluetoothGatt: registerApp() - UUID=5cca7371-d537-466c-ab47-488d6687c0c8
2021-11-09 15:07:47.284 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
2021-11-09 15:07:47.474 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=C4:EE:1A:38:C8:50
2021-11-09 15:07:47.477 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-11-09 15:07:47.478 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Connected to C4:EE:1A:38:C8:50
2021-11-09 15:07:47.483 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: wait(1600)
2021-11-09 15:07:48.130 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onConnectionUpdated() - Device=C4:EE:1A:38:C8:50 interval=6 latency=0 timeout=500 status=0
2021-11-09 15:07:48.132 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2021-11-09 15:07:48.480 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onConnectionUpdated() - Device=C4:EE:1A:38:C8:50 interval=39 latency=0 timeout=500 status=0
2021-11-09 15:07:48.481 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
2021-11-09 15:07:49.091 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.discoverServices()
2021-11-09 15:07:49.092 6269-6269/com.somewearlabs.sw.debug D/BluetoothGatt: discoverServices() - device: C4:EE:1A:38:C8:50
2021-11-09 15:07:49.101 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onSearchComplete() = Device=C4:EE:1A:38:C8:50 Status=0
2021-11-09 15:07:49.102 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Services discovered
2021-11-09 15:07:49.108 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.requestMtu(498)
2021-11-09 15:07:49.108 6269-6269/com.somewearlabs.sw.debug D/BluetoothGatt: configureMTU() - device: C4:EE:1A:38:C8:50 mtu: 498
2021-11-09 15:07:49.215 6269-12692/com.somewearlabs.sw.debug D/BluetoothGatt: onConfigureMTU() - Device=C4:EE:1A:38:C8:50 mtu=247 status=0
2021-11-09 15:07:49.218 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: MTU changed to: 247
2021-11-09 15:07:49.219 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.setCharacteristicNotification(da2e7828-fbce-4e01-ae9e-261174997c48, true)
2021-11-09 15:07:49.220 6269-6269/com.somewearlabs.sw.debug D/BluetoothGatt: setCharacteristicNotification() - uuid: da2e7828-fbce-4e01-ae9e-261174997c48 enable: true
2021-11-09 15:07:49.222 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2021-11-09 15:07:49.311 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2021-11-09 15:07:49.312 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Notifications enabled
2021-11-09 15:07:49.314 6269-6269/com.somewearlabs.sw.debug D/[main] FirmwareUpdater: onConnected
2021-11-09 15:07:49.329 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Sending (243 bytes) Header (Op: 2, Flags: 0, Len: 235, Group: 1, Seq: 0, Command: 1) CBOR {"data":"PbjzlgAAAAAAAgAAHCQGAAAAAAADBgsAAAAAAAAAAAD/////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////","len":403308,"sha":"Ikzf","off":0}
2021-11-09 15:07:49.331 6269-6269/com.somewearlabs.sw.debug D/[main] McuMgrBleTransport: gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48)
2021-11-09 15:07:49.333 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 02-00-00-EB-00-01-00-01-BF-64-64-61-74-61-58-CC-3D-B8-F3-96-00-00-00-00-00-02-00-00-1C-24-06-00-00-00-00-00-03-06-0B-00-00-00-00-00-00-00-00-00-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-FF-63-6C-65-6E-1A-00-06-27-6C-63-73-68-61-43-22-4C-DF-63-6F-66-66-00-FF
[... cut ...] ansport: Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-06-00-01-00-01-BF-62-72-63-02-FF
2021-11-09 15:07:49.415 6269-6269/com.somewearlabs.sw.debug I/[main] McuMgrBleTransport: Received Header (Op: 3, Flags: 0, Len: 6, Group: 1, Seq: 0, Command: 1) CBOR {"rc":2}
2021-11-09 15:07:49.439 6269-6269/com.somewearlabs.sw.debug D/[main] FirmwareUpdater: upgrade failed=UPLOAD; error=io.runtime.mcumgr.exception.McuMgrErrorException: Mcu Mgr Error: NO_MEMORY (2)
```

ayecarranda, 2021-11-10:
Also for reference I haven't had this issue with other devices, and the device should have enough memory

philips77, 2022-08-09:
Sorry for late reply. Seems like your device cannot save 403308 bytes.

This error may come from 
https://github.com/nrfconnect/sdk-zephyr/blob/482dcc7865f7d3a458f061a1d8276c5801eed1e4/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/img_mgmt.c#L451-L452
which returns ENOMEM here:
https://github.com/nrfconnect/sdk-zephyr/blob/482dcc7865f7d3a458f061a1d8276c5801eed1e4/subsys/mgmt/mcumgr/lib/cmd/img_mgmt/src/zephyr_img_mgmt.c#L345-L349

closed 2022-08-09 by philips77