nordicsemi/Android-nRF-Connect-Device-Manager #20: After upgrade completes, slots revert back to old firmware version

ayecarranda, 2021-11-10:
I see the device swap it's slots back (reverting to the old firmware image) after the library reports a successful upgrade. The library reports all stages of the upgrade succeeded (onUpgradeCompleted is invoked), but afterward, when the device is turned off and turned on again, the firmware reverts to the image prior to the upgrade. What could be causing this behavior? I can see the slots swap. I had this issue with 0.13.0-beta08 and also previously 0.11.0. You can see it says the 3.6.11 version has successfully swapped with 3.6.9, but then when I restart the device, the slots are swapped back to 3.6.9 as the primary and 3.6.11 as the secondary.

What is a workaround? 

Version: 0.13.0-beta08
Android Version: 10

Relevant log lines: 
```
2021-11-10 12:14:08.810 [main] DEBUG - onStateChanged: old=RESET, new=CONFIRM
2021-11-10 12:14:09.587 [main] DEBUG McuMgrBleTransport - [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2021-11-10 12:14:09.588 [main] WARN  McuMgrBleTransport - Error: (0x85): GATT ERROR
2021-11-10 12:14:09.589 [main] DEBUG McuMgrBleTransport - wait(500)
2021-11-10 12:14:10.094 [main] DEBUG McuMgrBleTransport - gatt.close()
2021-11-10 12:14:10.106 [main] DEBUG McuMgrBleTransport - wait(200)
2021-11-10 12:14:10.308 [main] TRACE McuMgrBleTransport - Retrying...
2021-11-10 12:14:10.311 [main] DEBUG McuMgrBleTransport - gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2021-11-10 12:14:10.852 [main] DEBUG McuMgrBleTransport - [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2021-11-10 12:14:10.854 [main] INFO  McuMgrBleTransport - Connected to F6:4E:EF:5E:A9:DB
2021-11-10 12:14:10.861 [main] DEBUG McuMgrBleTransport - wait(1600)
2021-11-10 12:14:11.507 [main] INFO  McuMgrBleTransport - Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
2021-11-10 12:14:11.862 [main] INFO  McuMgrBleTransport - Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
2021-11-10 12:14:12.467 [main] TRACE McuMgrBleTransport - Discovering services...
2021-11-10 12:14:12.469 [main] DEBUG McuMgrBleTransport - gatt.discoverServices()
2021-11-10 12:14:12.480 [main] INFO  McuMgrBleTransport - Services discovered
2021-11-10 12:14:12.482 [main] TRACE McuMgrBleTransport - Primary service found
2021-11-10 12:14:12.487 [main] TRACE McuMgrBleTransport - Requesting new MTU...
2021-11-10 12:14:12.487 [main] DEBUG McuMgrBleTransport - gatt.requestMtu(498)
2021-11-10 12:14:12.594 [main] INFO  McuMgrBleTransport - MTU changed to: 247
2021-11-10 12:14:12.597 [main] DEBUG McuMgrBleTransport - gatt.setCharacteristicNotification(da2e7828-fbce-4e01-ae9e-261174997c48, true)
2021-11-10 12:14:12.600 [main] TRACE McuMgrBleTransport - Enabling notifications for da2e7828-fbce-4e01-ae9e-261174997c48
2021-11-10 12:14:12.602 [main] DEBUG McuMgrBleTransport - gatt.writeDescriptor(00002902-0000-1000-8000-00805f9b34fb, value=0x01-00)
2021-11-10 12:14:12.690 [main] INFO  McuMgrBleTransport - Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
2021-11-10 12:14:12.693 [main] INFO  McuMgrBleTransport - Notifications enabled
2021-11-10 12:14:12.695 [main] DEBUG FirmwareUpdater - onConnected
2021-11-10 12:14:12.700 [main] INFO  McuMgrBleTransport - Sending (58 bytes) Header (Op: 2, Flags: 0, Len: 50, Group: 1, Seq: 0, Command: 0) CBOR {"confirm":true,"hash":"cYQV7cnhpbmWnj1Jf7ZWqsIDb8A3UMoFaQagI+jLBU8="}
2021-11-10 12:14:12.701 [main] TRACE McuMgrBleTransport - Writing characteristic da2e7828-fbce-4e01-ae9e-261174997c48 (WRITE COMMAND)
2021-11-10 12:14:12.702 [main] DEBUG McuMgrBleTransport - gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48)
2021-11-10 12:14:12.706 [main] INFO  McuMgrBleTransport - Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 02-00-00-32-00-01-00-00-BF-67-63-6F-6E-66-69-72-6D-F5-64-68-61-73-68-58-20-71-84-15-ED-C9-E1-A5-B9-96-9E-3D-49-7F-B6-56-AA-C2-03-6F-C0-37-50-CA-05-69-06-A0-23-E8-CB-05-4F-FF
2021-11-10 12:14:12.789 [main] INFO  McuMgrBleTransport - Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-87-00-01-00-00-BF-66-69-6D-61-67-65-73-9F-BF-64-73-6C-6F-74-00-67-76-65-72-73-69-6F-6E-66-33-2E-36-2E-31-31-64-68-61-73-68-58-20-71-84-15-ED-C9-E1-A5-B9-96-9E-3D-49-7F-B6-56-AA-C2-03-6F-C0-37-50-CA-05-69-06-A0-23-E8-CB-05-4F-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F5-66-61-63-74-69-76-65-F5-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-FF-6B-73-70-6C-69-74-53-74-61-74-75-73-00-FF
2021-11-10 12:14:12.795 [main] INFO  McuMgrBleTransport - Received Header (Op: 3, Flags: 0, Len: 135, Group: 1, Seq: 0, Command: 0) CBOR {"images":[{"slot":0,"version":"3.6.11","hash":"cYQV7cnhpbmWnj1Jf7ZWqsIDb8A3UMoFaQagI+jLBU8=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}
2021-11-10 12:14:12.808 [main] TRACE Confirm - Confirm response: {"images":[{"slot":0,"version":"3.6.11","hash":"cYQV7cnhpbmWnj1Jf7ZWqsIDb8A3UMoFaQagI+jLBU8=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}
2021-11-10 12:14:12.809 [main] DEBUG - onUpgradeCompleted
```

philips77, 2022-03-23:
Hello,
Sorry for the very late response. I must have missed your issue. Did you manage to solve it?

philips77, 2022-08-09:
The primary slot may be reverted to the initial firmware if the uploaded image was invalid (e.g. signed with a wrong signature). This is intended behavior to protect the original image.

closed 2022-08-09 by philips77