nordicsemi/Android-nRF-Connect-Device-Manager #37: Using different SMP service and characteristics UUID for McuMgrBleTransport

marianhlavac, 2022-02-10:
What is considered to be the best approach to choosing different SMP service and characteristics UUID for the _McuMgrBleTransport_ instance? Forgive me if I'm wrong, but there doesn't seem to be any sensible mechanism in the code for extending the current _McuMgrBleTransport_ with UUIDs overrides. 

The only option seems to be replacing the whole _McuMgrGattCallback_ class in the `CustomMcuMgrBleTransport.getGattCallback()`, which contains many references to internal structure of the _McuMgrBleTransport_, such as `mSmpCharacteristicNotify`, `mSmpCharacteristicWrite`, `mSmpProtocol` and so on, so as a result it's not an option at all. Our goal is not to write completely custom transport or completely custom GattCallback. We would like to use the exactly same logic that is already present in the _McuMgrBleTransport_, just with different UUIDs.

Is there any reason for such strict access control in these methods and subclasses? Our device has exposed two SMP services (we have two nRF chips on board) and we need to access both SMP services with single mobile application. Unfortunately we're currently not able to do this without very ugly hacks and code duplication, which also effectively prevents us from easily updating to the latest library version.

The most appreciated solution for this would be parametrized constructor of McuMgrTransport, which would allow us to choose different UUIDs. Is this possible to include with the library?

###### (Sorry for making this question an issue, I'd rather use the discussions feature, but it's disabled for this repo)

philips77, 2022-02-10:
Sure. I'll add it to the final release. Please, expect it on 1.1-beta02 tomorrow or early next week.

Btw, why do you want to use custom UUID instead of the default one?

marianhlavac, 2022-02-17:
Sorry for being pushy – is there any updates for the beta.2? Can we offer help with something?

philips77, 2022-02-17:
Not yet, but I remember!
I got distracted with something else. Sorry!

closed 2022-03-23 by philips77