nordicsemi/Android-nRF-Connect-Device-Manager #51: nordic chip is not being disconnect after , ncs firmware update completes

divya-chaubey, 2022-05-05:
we are updating firmware using this library in our nordic chip. but after successful firmware update completes , nordic chip is still connected with our phone . but this case does not happen in ios version of library. can you please brief what should we do , so that chip automatically get disconnects from phone once firmware update completes as like ios .

philips77, 2022-05-05:
Hello,
Which version of the app are you talking about?
What upgrade mode are you using (*test only*, *confirm only*, *test and confirm*)?
What if you turn off the "Erase application settings" switch?

divya-chaubey, 2022-05-05:
> <NordicSemiconductor/Android-nRF-Connect-Device-Manager/issues/51/1118527330
> @github.com>

We are using 1.2.0 version of the library.

Upgrade mode using-> Confirm only

We have tried with both "Erase Application settings" true as wella s false
. It is still remains connected after firmware update completes

divya-chaubey, 2022-05-06:
> That is weird. In *confirm only* mode the library sends Reset command at
> the end, which should reboot the device triggering the mcu boot to swap
> images. Do you have logs from LogCat?

Hello,

I am attaching a screenshot of logcat that I am receiving after
successful firmware update. Our android app is still connected to a
peripheral device (Nordic semiconductor chip).
Upgrade mode -> Confirm Only
Erase Application settings -> true

Kindly look into this.

*Thanks & Regards,*
*- Divya Chaubey*
*Associate Developer*

*Agicent Technologies Pvt. Ltd.*
www.agicent.com

philips77, 2022-05-07:
Hi, please, attach the file on GitHub, not via email. Email attachments don't work here.

gulluchaubey, 2022-05-11:
> Hi, please, attach the file on GitHub, not via email. Email attachments don't work here.
[image]

philips77, 2022-05-11:
In the first step of DFU process the library is sending "Image read" command to find out what exactly is to be done. It compares the returned hashes with those from the provided file and determined which cores have to be updated. If all hashes match, it immediately completes with status success.

closed 2022-08-09 by philips77

divya-chaubey, 2022-10-11:
One more point I would like to add. After firmware update completes, device
gets reset and then reconnect. Please look into this.

philips77, 2022-10-11:
What upload mode are you using? With "Confirm only" it should not reconnect.