nordicsemi/Android-nRF-Connect-Device-Manager #61: Version 1.2.0 does not detect any devices on Android 12 Patch May 5

simonratner, 2022-05-30:
Scanning does not detect any nearby devices, this used to work previously with the same hardware device.

I notice that the app is no longer asking for Location permission, and does not have it granted in the app settings page. The only permission it has is "Nearby Devices", not sure if an explicit "Location" permission is still required in this Android version. Could this be an issue?

For comparison, the nRF Connect app does have the Location permission, and works fine (detects same device, including the SMP service).

simonratner, 2022-05-30:
As an aside: is it possible to download an official apk of previous versions from somewhere?

philips77, 2022-05-31:
> Scanning does not detect any nearby devices, this used to work previously with the same hardware device.

This should work. Try restarting Bluetooth or the phone. Perhaps erasing app's storage, etc.

> I notice that the app is no longer asking for Location permission, and does not have it granted in the app settings page. The only permission it has is "Nearby Devices", not sure if an explicit "Location" permission is still required in this Android version. Could this be an issue?

On Android 12+ the location permission is no longer required if one scans with `neverForLocation` flag. [Read more](https://developer.android.com/guide/topics/connectivity/bluetooth/permissions)

> For comparison, the nRF Connect app does have the Location permission, and works fine (detects same device, including the SMP service).

Yes, because in nRF Connect we also want to show *iBeacons* and *Eddystone beacons*, and to scan those the location permission is required. Unfortunately, it is not possible to ask user to choose (scan with beacons or without beacons), as this has to be decided on the compilation time and without the location permission we're not getting scan results if we don't set this `neverForLocation` flag. Bad design decision in Android.

> As an aside: is it possible to download an official apk of previous versions from somewhere?

I recommend, actually, fixing the scanning issue on the latest version. We're also testing on Android 12 and everything works smoothly.