nordicsemi/Android-nRF-Connect-Device-Manager #68: Firmware Upgrade "Connection timed out" in 1.4.0

gte1, 2022-07-01:
Using smp_svr example from nRFConnect SDK 2.0.0, with overlay-bt.conf (which should have improved DFU speed)
Android 10 (Huawei P20 pro), nRF52840dk

Expected: Firmware upgrade works
Actual: Firmware upgrade "Connection timed out"

philips77, 2022-07-02:
Hello,
Please post here relevant logs that you get from the manager or BLE transport.

gte1, 2022-07-04:
I am using the nRFConnect Device Manager 1.4.0 from Google Play. Is there any way to retrieve logs from that one? I cannot finda menu item for logs.

gte1, 2022-07-04:
Just to inform. We tried again using another phone, Google Pixel 6 with Android 12, and saw no problems there (and got a very fast firmware update speed)
So maybe the problem is that the Device Manager app is not completely compatible with older Android 10 and/or Huawei P20 pro

philips77, 2022-07-04:
> Is there any way to retrieve logs from that one?

You may install nRF Logger app and you'll see the logs there.

gte1, 2022-07-04:
Log from Android 10 phone
`Device Manager, 2022-07-04
Zephyr (Device Manager)
V	14:40:57.176	[j] Starting DFU, mode: TEST_AND_CONFIRM
I	14:40:57.770	[e] Connected to D4:12:A7:99:E8:BD
I	14:40:58.088	[e] Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I	14:40:58.156	[e] PHY updated (TX: LE 2M, RX: LE 2M)
I	14:40:58.310	[e] Services discovered
I	14:40:58.373	[e] Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
I	14:40:58.423	[e] MTU changed to: 498
I	14:40:58.519	[e] Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
I	14:40:58.531	[e] Notifications enabled
I	14:40:58.545	[e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-01-00-00-00-06-A0
I	14:40:58.615	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-19-00-00-00-06-BF-68-62-75-66-5F-73-69-7A-65-19-09-AB-69-62-75-66-5F-63-6F-75-6E-74-04-FF
I	14:40:58.640	[e] Received Header (Op: 1, Flags: 0, Len: 25, Group: 0, Seq: 0, Command: 6) CBOR {"buf_size":2475,"buf_count":4}
I	14:40:58.682	[e] SMP reassembly supported with buffer size: 2475 bytes and count: 4
W	14:40:58.689	[e] PHY read failed with status 6
I	14:40:58.708	[e] Sending (10 bytes) Header (Op: 0, Flags: 0, Len: 2, Group: 1, Seq: 0, Command: 0) CBOR {}
I	14:40:58.911	[e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-02-00-01-00-00-BF-FF
I	14:40:58.927	[e] Sending (10 bytes) Header (Op: 0, Flags: 0, Len: 2, Group: 0, Seq: 1, Command: 6) CBOR {}
I	14:40:58.942	[e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-02-00-00-01-06-BF-FF
I	14:40:59.005	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-86-00-01-00-00-BF-66-69-6D-61-67-65-73-9F-BF-64-73-6C-6F-74-00-67-76-65-72-73-69-6F-6E-65-30-2E-30-2E-30-64-68-61-73-68-58-20-5C-F4-DA-8F-47-29-E9-6D-11-3F-0A-FB-59-72-EE-3C-DC-89-D8-38-A2-C2-E7-6C-48-A5-6A-71-0E-55-70-04-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F5-66-61-63-74-69-76-65-F5-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-FF-6B-73-70-6C-69-74-53-74-61-74-75-73-00-FF
I	14:40:59.029	[e] Received Header (Op: 1, Flags: 0, Len: 134, Group: 1, Seq: 0, Command: 0) CBOR {"images":[{"slot":0,"version":"0.0.0","hash":"XPTaj0cp6W0RPwr7WXLuPNyJ2DiiwudsSKVqcQ5VcAQ=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}
V	14:40:59.071	[k] Validation response: {"images":[{"slot":0,"version":"0.0.0","hash":"XPTaj0cp6W0RPwr7WXLuPNyJ2DiiwudsSKVqcQ5VcAQ=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false}],"splitStatus":0}
V	14:40:59.087	[j] Moving from state VALIDATE to state UPLOAD
I	14:40:59.099	Uploading firmware...
W	14:41:00.144	[Uploader] Uploader write failure for chunk with offset=0: Transaction 2 timed out without receiving a response
W	14:41:01.190	[Uploader] Uploader write failure for chunk with offset=0: Transaction 3 timed out without receiving a response
W	14:41:02.229	[Uploader] Uploader write failure for chunk with offset=0: Transaction 4 timed out without receiving a response
W	14:41:03.253	[Uploader] Uploader write failure for chunk with offset=0: Transaction 5 timed out without receiving a response
W	14:41:04.278	[Uploader] Uploader write failure for chunk with offset=0: Transaction 6 timed out without receiving a response
E	14:41:04.301	[ImageUploader] Upload failed
Transaction 6 timed out without receiving a response
	at r2.e$b.b(Unknown Source:20)
	at s2.f.h(Unknown Source:10)
	at s2.f.b(Unknown Source:0)
	at s2.c.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:888)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:213)
	at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: Transaction 6 timed out without receiving a response
	at s2.b$h.j(Unknown Source:76)
	at b4.a.r(Unknown Source:11)
	at p4.r0.run(Unknown Source:88)
	at kotlinx.coroutines.scheduling.a.l(Unknown Source:0)
	at kotlinx.coroutines.scheduling.a$c.d(Unknown Source:14)
	at kotlinx.coroutines.scheduling.a$c.n(Unknown Source:28)
	at kotlinx.coroutines.scheduling.a$c.run(Unknown Source:0)

E	14:41:04.316	Upgrade failed
Transaction 6 timed out without receiving a response
	at r2.e$b.b(Unknown Source:20)
	at s2.f.h(Unknown Source:10)
	at s2.f.b(Unknown Source:0)
	at s2.c.run(Unknown Source:4)
	at android.os.Handler.handleCallback(Handler.java:888)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:213)
	at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: Transaction 6 timed out without receiving a response
	at s2.b$h.j(Unknown Source:76)
	at b4.a.r(Unknown Source:11)
	at p4.r0.run(Unknown Source:88)
	at kotlinx.coroutines.scheduling.a.l(Unknown Source:0)
	at kotlinx.coroutines.scheduling.a$c.d(Unknown Source:14)
	at kotlinx.coroutines.scheduling.a$c.n(Unknown Source:28)
	at kotlinx.coroutines.scheduling.a$c.run(Unknown Source:0)

I	14:41:08.808	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-0D-00-01-02-01-BF-62-72-63-00-63-6F-66-66-19-09-80-FF
I	14:41:08.823	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-0D-00-01-03-01-BF-62-72-63-00-63-6F-66-66-19-09-80-FF
[... cut ...]
I	14:41:14.059	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-0D-00-01-05-01-BF-62-72-63-00-63-6F-66-66-19-09-80-FF
I	14:41:14.070	[e] Received Header (Op: 3, Flags: 0, Len: 13, Group: 1, Seq: 5, Command: 1) CBOR {"rc":0,"off":2432}
I	14:41:15.178	[e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-0D-00-01-06-01-BF-62-72-63-00-63-6F-66-66-19-09-80-FF
I	14:41:15.191	[e] Received Header (Op: 3, Flags: 0, Len: 13, Group: 1, Seq: 6, Command: 1) CBOR {"rc":0,"off":2432}
I	14:41:19.384	[e] Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 420ms)
`

hubpav, 2022-07-05:
The `Connection Timeout` message in the nRF Device Manager app happens with my Google Pixel 3a XL as well. Repeated attempts to upload the image will succeed.

```
Device Manager, 2022-07-05
CHESTER 2155872512 (Device Manager)
V 22:00:19.505 [j] Starting DFU, mode: TEST_AND_CONFIRM
I 22:00:21.535 [e] Connected to D1:2B:96:3E:6F:17
I 22:00:21.945 [e] Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I 22:00:22.426 [e] Services discovered
I 22:00:22.456 [e] MTU changed to: 247
I 22:00:22.486 [e] Data written to descr. 00002902-0000-1000-8000-00805f9b34fb, value: (0x) 01-00
I 22:00:22.493 [e] Notifications enabled
I 22:00:22.503 [e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-01-00-00-00-06-A0
I 22:00:22.532 [e] Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
I 22:00:22.624 [e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-06-00-00-00-06-BF-62-72-63-08-FF
I 22:00:22.643 [e] Received Header (Op: 1, Flags: 0, Len: 6, Group: 0, Seq: 0, Command: 6) CBOR {"rc":8}
I 22:00:22.650 [e] PHY read (TX: LE 2M, RX: LE 2M)
I 22:00:22.661 [e] Sending (10 bytes) Header (Op: 0, Flags: 0, Len: 2, Group: 1, Seq: 0, Command: 0) CBOR {}
I 22:00:22.867 [e] Sending (10 bytes) Header (Op: 0, Flags: 0, Len: 2, Group: 0, Seq: 1, Command: 6) CBOR {}
I 22:00:22.877 [e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-02-00-01-00-00-BF-FF
I 22:00:22.886 [e] Data written to da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 00-00-00-02-00-00-01-06-BF-FF
I 22:00:22.939 [e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-F4-00-01-00-00-BF-66-69-6D-61-67-65-73-9F-BF-64-73-6C-6F-74-00-67-76-65-72-73-69-6F-6E-65-30-2E-30-2E-30-64-68-61-73-68-58-20-6C-11-52-23-8B-79-4D-F3-47-CC-AA-EF-A5-4F-49-35-D7-05-AC-D0-ED-3D-5C-C9-33-9D-72-EB-12-96-B3-53-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F5-66-61-63-74-69-76-65-F5-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-BF-64-73-6C-6F-74-01-67-76-65-72-73-69-6F-6E-65-30-2E-30-2E-30-64-68-61-73-68-58-20-67-8D-5A-9C-C2-49-23-06-D2-3A-7F-8D-46-84-54-F7-46-78-58-E6-FA-D1-28-BB-6C-E6-A1-9D-61-6D-07-E1-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F4-66-61-63-74-69-76-65-F4-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-FF-6B-73-70-6C-69-74
I 22:00:22.952 [e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 53-74-61-74-75-73-00-FF
I 22:00:22.958 [e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-06-00-00-01-06-BF-62-72-63-08-FF
I 22:00:22.966 [e] Received Header (Op: 1, Flags: 0, Len: 244, Group: 1, Seq: 0, Command: 0) CBOR {"images":[{"slot":0,"version":"0.0.0","hash":"bBFSI4t5TfNHzKrvpU9JNdcFrNDtPVzJM51y6xKWs1M=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false},{"slot":1,"version":"0.0.0","hash":"Z41anMJJIwbSOn+NRoRU90Z4WOb60Si7bOahnWFtB+E=","bootable":true,"pending":false,"confirmed":false,"active":false,"permanent":false}],"splitStatus":0}
I 22:00:22.973 [e] Received Header (Op: 1, Flags: 0, Len: 6, Group: 0, Seq: 1, Command: 6) CBOR {"rc":8}
V 22:00:22.988 [k] Validation response: {"images":[{"slot":0,"version":"0.0.0","hash":"bBFSI4t5TfNHzKrvpU9JNdcFrNDtPVzJM51y6xKWs1M=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false},{"slot":1,"version":"0.0.0","hash":"Z41anMJJIwbSOn+NRoRU90Z4WOb60Si7bOahnWFtB+E=","bootable":true,"pending":false,"confirmed":false,"active":false,"permanent":false}],"splitStatus":0}
V 22:00:22.993 [j] Moving from state VALIDATE to state UPLOAD
I 22:00:23.000 Uploading firmware...
W 22:00:24.013 [Uploader] Uploader write failure for chunk with offset=0: Transaction 2 timed out without receiving a response
W 22:00:25.030 [Uploader] Uploader write failure for chunk with offset=0: Transaction 3 timed out without receiving a response
W 22:00:26.047 [Uploader] Uploader write failure for chunk with offset=0: Transaction 4 timed out without receiving a response
W 22:00:27.061 [Uploader] Uploader write failure for chunk with offset=0: Transaction 5 timed out without receiving a response
W 22:00:28.073 [Uploader] Uploader write failure for chunk with offset=0: Transaction 6 timed out without receiving a response
E 22:00:28.086 [ImageUploader] Upload failed
Transaction 6 timed out without receiving a response
at r2.e$b.b(Unknown Source:20)
at s2.f.h(Unknown Source:10)
at s2.f.b(Unknown Source:0)
at s2.c.run(Unknown Source:4)
at android.os.Handler.handleCallback(Handler.java:938)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loopOnce(Looper.java:201)
at android.os.Looper.loop(Looper.java:288)
at android.os.HandlerThread.run(HandlerThread.java:67)
Caused by: Transaction 6 timed out without receiving a response
at s2.b$h.j(Unknown Source:76)
at b4.a.r(Unknown Source:11)
at p4.r0.run(Unknown Source:88)
at kotlinx.coroutines.scheduling.a.l(Unknown Source:0)
at kotlinx.coroutines.scheduling.a$c.d(Unknown Source:14)
at kotlinx.coroutines.scheduling.a$c.n(Unknown Source:28)
at kotlinx.coroutines.scheduling.a$c.run(Unknown Source:0)

E 22:00:28.095 Upgrade failed
Transaction 6 timed out without receiving a response
at r2.e$b.b(Unknown Source:20)
at s2.f.h(Unknown Source:10)
at s2.f.b(Unknown Source:0)
at s2.c.run(Unknown Source:4)
at android.os.Handler.handleCallback(Handler.java:938)
[... cut ...]
I 22:00:32.961 [e] Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 03-00-00-0C-00-01-04-01-BF-62-72-63-00-63-6F-66-66-18-CC-FF
I 22:00:32.965 [e] Received Header (Op: 3, Flags: 0, Len: 12, Group: 1, Seq: 2, Command: 1) CBOR {"rc":0,"off":204}
I 22:00:32.971 [e] Received Header (Op: 3, Flags: 0, Len: 12, Group: 1, Seq: 3, Command: 1) CBOR {"rc":0,"off":204}
I 22:00:32.975 [e] Received Header (Op: 3, Flags: 0, Len: 12, Group: 1, Seq: 4, Command: 1) CBOR {"rc":0,"off":204}
I 22:00:33.092 [e] Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 4000ms)
```

gte1, 2022-07-06:
**Samsung Galaxy S9, Android 10: Fails on first try, but success on retry**

[Log 2022-07-06 11_15_59.txt](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/files/9053469/Log.2022-07-06.11_15_59.txt)

philips77, 2022-07-06:
Try sending erase command before uploading the image.
I don't know why

philips77, 2022-07-06:
You may also create a PR.
In the `Validate` task in DFU performer and `Erase` task if hashes don't match.

gte1, 2022-07-06:
Erase command works as you proposed on Galaxy S9, Android 10
On Huawei P20 Pro, Android 10, this did not help, probably due to the "PHY read failed with status 6" issue
I am a firmware developer and have no experience with Android development, so a PR from me will probably not happen.

gte1, 2022-07-06:
> Try sending erase command before uploading the image.
> 
> As the slot it full, the first packet with fw will erase the slot. This takes time, up to 30 sec. But I'm version 1.4.0 we decreased the timeout for upload command to 1 second. I'll fix it in the next version.

Maybe this timeout can be a setting in the app?

philips77, 2022-07-06:
There are 3 options:
- a timeout for image upload may be increased to 30 sec, like it was before (but that was causing another issues)
- a timeout for the first image upload command, where off=0, can be set to 30 sec, but that's unclear and would have to be documented
- a new `Erase` command added to erase the slot if needed.

gte1, 2022-07-06:
Huawei P20 LITE, Android 9, works in advanced mode (with manual Erase step), even if this device also logs "PHY read failed with status 6". Approx 7 KB/s. Sometimes, I have to retry clicking Upload button, if "Connection timed out" message appeared

gte1, 2022-07-07:
But that 'connect method' would not be accessible if using the sample app from Google Play?

philips77, 2022-07-10:
Right, but there you can simply connect by sending Hello echo command, erase using Advanced mode on Image tab.

gte1, 2022-07-12:
I did that on Huawei P20 PRO, Android 10. I still get "Connection timed out", even if I pre-connected via sending Hello.
Slot 1, was already erased from earlier attempt.

See log
[Log 2022-07-12 09_13_20.txt](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/files/9090498/Log.2022-07-12.09_13_20.txt)

gte1, 2022-07-12:
On Huawei P20 **Lite**, Android 10, pre-connecting did not help. Still have to retry there after getting "Connection timed out", but succeeds after retry.

philips77, 2022-08-09:
Hi,
I've started working on this lib and will add the `Erase` command. Also, have a look at #70. Perhaps that fixed your issue?

closed 2022-08-09 by philips77

gte1, 2022-08-09:
I will check as soon as 1.4.1 is on Google Play

PR #71 (2022-08-09): Connection timeout fixed
This PR fixes #68. When the secondary slot of some image is not-empty, but hasn't been confirmed or tested, and user tries to upload a different image, the slot will be automatically erased.
However, when progressive flash or stream flash are not turned on, this may take some time, depending on the flash hardware. This PR sets the timeout for the first upload chunk (where offset = 0) to 30 seconds (was 1 second before).

This approach is better then adding an `Erase` command before upload, like proposed in #68, as `Erase` always takes a long time (erases whole slot), w [... cut]