nordicsemi/Android-nRF-Connect-Device-Manager #73: nRF9160 Mcumgr FMFU update support

optical-o, 2022-08-16:
Hello, 
i would like to ask you, whether you are planing to implement the FMFU part of the Nordic Connect SDK into the nRF connect device manager application. 

- The sdk is supporting this by custom nordic implementation, which is not public. It is distributed as closed source DLL. 
- Due to the release of modem firmware 1.3.2, we will be forced to update the firmware on our devices, yet we are still missing open implementation for this. We would be also satisfied with closed one integrated into nrf connect device manager.
- We have verified the functionality that our mcumgr inside the device is capable of receiving the fmfu by using other transport layer(UART). However, we would like to have ability to update the modem firmware from the mobile.
- I think we are definitely not alone who will face this issue, with upcoming releases of the modem firmware.

[FMFU](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.0.2/nrf/libraries/dfu/fmfu_mgmt.html)

philips77, 2022-08-17:
Do you have a Bluetooth LE proxy in the nRF91 device? As far as I know, DFU for 91 devices is done using cellular.

optical-o, 2022-08-17:
Yes, we have Bluetooth LE connectivity by using external Bluetooth controller (nRF52833).

optical-o, 2022-08-17:
The DFU using cellular is very data intensive task. Our devices have very limited budget. We already have integrated this library into our mobile application for regular DFU. We would like to extend the update process with a FMFU step.

optical-o, 2022-08-17:
The verification that the FMFU is working correctly using:
```
from pynrfjprog import HighLevel

with HighLevel.API() as api:
    with HighLevel.ModemUARTDFUProbe(api, uart, baudrate) as modem_dfu_probe:
        modem_dfu_probe.program(modem_firmware_zip)
        modem_dfu_probe.verify(modem_firmware_zip)
```

If i understand correctly that the endpoints should be available using the same mcumgr transport layer. The Logic of parsing and sending mfw data to the endpoints is closed inside the DLL nrfjprog.

philips77, 2022-08-17:
Thank you for the answer. I'll talk with my colleagues and come back to you. I have no knowledge about FMFU yet.
McuMgr doesn't care about the transport, so BLE should work.

philips77, 2022-08-18:
Btw, RFC for tunelling SMP commands via other device: https://github.com/zephyrproject-rtos/zephyr/issues/44224
May I ask how are you updating the main fw on 91?

optical-o, 2022-08-18:
We are using the nrf91 as Bluetooth Host. So the bluetooth connection is directly interfacing with the nrf91 chip.

The question is probably targeted on how we are updating the nrf52 controller. The answer is second GATT service with different UUID, which forwards the requests to the nrf52 mcumgr over our multiplexed UART transport layer(HCI interface, Mcumgr interface). The tunneling feature would allow us to ditch the secondary UUID. This would make our solution a little neater.  Our solution is by no means perfect.

optical-o, 2022-09-26:
@philips77  Did you manage to get information about the FMFU Mcumgr implementation ?

We would like to be able upgrade our devices to modem firmware 1.3.2. First thing when we are raising tickets on devzone is response that we should retest with the newest available modem firmware, which we are unable to do.

VVAQARAHMED, 2023-04-04:
Hi, I am in the same situation i want to upgrade the modem firmware FMFU(nrf91) of remote devices.
a couple of option i tried although i am successful  in dev. downloading  modem firmware into sd card and then updating from sd card.I had successfully done it in dev environment .
our firmware  have Zephyr Shell capability and I am trying to upgrade the firmware through UART(FTDI cable).
i am able to update the application through mcumgr but not modem firmware