nordicsemi/Android-nRF-Connect-Device-Manager #74: FS module download/upload speed

optical-o, 2022-08-29:
Hi,
i would like to ask you, if there is plan to focus on download speed for the Mcumgr FS Module. The download speed of this module is rather slow. Do you experience similar issue ? 

We are downloading about 0.5MB of logs from the device and it takes ages. I guess the similar issue will be when anyone uploads files to the device(That is another problem issue). We would like to be able to download files from the device in reasonable times.

Is there a plan to create similar window download method as for the Image upload ?

philips77, 2022-08-29:
I've asked our SDK team, which is developing the firmware. I'm not aware of any plans, but let's see.
We have optimized DFU upload speed recently, so the code is there, just needs to be copied to the right places, I believe. But that requires changes on the Zephyr and mobile side.

philips77, 2022-11-24:
OK, as far as I know, buffering is done on transport layer on the device side, so this should work for any group, including FS. I have to look into current API in this library.

PR #85 (2022-11-30): FS upload speed improvement
This PR fixes #74.

The new `Uploader` mechanism has been made more generic so it can be used by other managers besides `ImageManager`.

See c6f50218c74df9aa08aec0f6b79cd05c791db128 for migration guide.
closed 2022-11-30 by philips77