nordicsemi/Android-nRF-Connect-Device-Manager #75: [Feature] Make the library bindings-friendly for the sake of C# / Xamarin

ksidirop-laerdal, 2022-09-01:
Currently there are some severe problems when it comes to binding kotlin stuff over to C#. To my knowledge there's no official C# / Xamarin bindings project for this library. Maybe you could take up on this side-quest and introduce such a project? It would definately be highly appreciated.

bravekit, 2022-11-18:
I am also very interested in porting libraries to C# for using them with Xamarin or MAUI

philips77, 2022-11-19:
We have no C# or Xamarin experience, so I'm sorry, but we can't help much. If you need, please create such projects and we'll gladly point to them in the Readme.

philips77, 2022-11-19:
Also, i don't know what's bindings-friendly mean in this context. If you need and changes, prepare a PR and we'll help you merge it.

ksidirop-laerdal, 2023-09-01:
> Also, i don't know what's bindings-friendly mean in this context. If you need and changes, prepare a PR and we'll help you merge it.

Let me give you a quick practical example. In my team we created C# nugets that rely on java-bindings under the surface to interact with Nordic BLE devices. Up to version 1.5.1 of your libs everything was working smoothly. In the recent months however your libs started using 'sealed' classes:

https://github.com/search?q=repo%3ANordicSemiconductor%2FAndroid-nRF-Connect-Device-Manager%20sealed&type=code 

When we upgraded the underlying java libs to version 1.7.2 we realized that even though our nuget packages where getting properly built (without any errors), when we tried to install+build "Classic Xamarin" apps for Android we would get a build-error like this one:

[image]

We tried to fork + refactor Nordic-Android-McuMgr ver 1.7.2 so that it won’t rely on the ‘sealed’ keyword but even when our wrapper was compiled against this modified version of the Nordic-Android-McuMgr library there was still an issue: The Sample App would compile and run with the new nugets but when we would try to perform a fw installation we would get a runtime error during the upload phase (could be related to the changes I made in those ‘sealed’ classes - not sure though - I'm still looking into this)

As far as I looked into this either

(a) the language features of your java libs have to be conciously kept compatible with the "Classic Xamarin" toolchain  (so no more 'sealed')

or (b) we will have to freeze support for "Classic Xamarin" and focus our aim to .Net6+ because only the updated r8 provided by the .Net6+ toolchain can cope with stuff like 'sealed' and so on - but even in this case if you keep using new java language features at will chances are that the C# toolchain will break again in random ways at some point thus forcing us to constantly support only the most bleeding-edge versions of .Net which is far from ideal in terms of adoption.

>  If you need, please create such projects and we'll gladly point to them in the Readme.

We will consider it - much appreciated! Keep up the good work!

ksidirop-laerdal, 2023-10-27:
> I am also very interested in porting libraries to C# for using them with Xamarin or MAUI

You're in luck - the company I'm working for is planning on open-sourcing our C# wrapper-nugets on Nordic native libs for both Xamarin and MAUI (.net6+) encompassing iOS, Android and MacCatalyst (but not Windows - that would require a total rewrite of the libs to C#). So stay tuned!

@bravekit

closed 2023-10-27 by ksidirop-laerdal

bravekit, 2023-10-27:
> You're in luck - the company I'm working for is planning on open-sourcing our C# wrapper-nugets on Nordic native libs for both Xamarin and MAUI (.net6+) encompassing iOS, Android and MacCatalyst (but not Windows - that would require a total rewrite of the libs to C#). So stay tuned!
> 
That's super cool! Where can I follow your progress?

ksidirop-laerdal, 2023-10-27:
> That's super cool! Where can I follow your progress?

Everything is still under wraps - but we will make an announcement in due time when our codebase goes public. Will let you know @bravekit

giacomo-cavicchioli-zupit, 2024-06-07:
@ksidirop-laerdal hi, any news on the nuget package? we trying to integrate these nordic libraries in a Xamarin (and in a few months MAUI) app