nordicsemi/Android-nRF-Connect-Device-Manager #76: Pause() and Resume() image upload from different devices

zhamaliddin-kalzhan, 2022-09-06:
Hello, I am using Mcumgr for DFU on nRF52832DK.
I wanted to implement robust DFU, so that if the connection is suddenly terminated during a session with one device, it can be resumed on the consequent connection from another device.

The README file says pause() and resume() is supported only for window capacity set to 1.
However, while testing I noticed that if bonding information is stored, the upload can be resumed on reconnection with the same device even when window capacity set higher(3).

Is this feature also supported for pausing image upload from one and resume from another device?

philips77, 2022-09-06:
Some recent version added resuming upload and pause/resume to the uploader, which means it's now supported also with window capacity > 1.

I will update the doc.

philips77, 2022-11-30:
Fixed.

closed 2022-11-30 by philips77