nordicsemi/Android-nRF-Connect-Device-Manager #78: FS Upload file

optical-o, 2022-10-12:
Hello,
I'm having issue when uploading larger files onto the device:

[image]

By the error prompt it seems that the FS layer is trying to send larger chunk than supported MTU.

optical-o, 2022-10-12:
The device is using the SMP Reassembly.

philips77, 2022-10-12:
Hi, this seems like a bug in the library. Before sending a packet it calculates is size, and this calculations looks like off by 1. I'll have a look at it soon.

optical-o, 2022-10-14:
I have seen even Payload (2477) so it might be more than off by 1.

optical-o, 2022-10-25:
@philips77  Could you give us a approximate date when you get to the issue ?
Thank you

JeffWelder, 2022-11-01:
Hi there. 
I got a "Payload (497 bytes) too long for MTU: 495" if that's any help when you get around to this. 
Many thanks for this tool, it has helped a ton with development. :)

philips77, 2022-11-24:
Hi,
Quick question. The above mentioned error doesn't happen at the beginning of a file, but rather later, is that right?
And does it work for small files, but fails when the `offset` reaches some point, always the same value?

philips77, 2022-11-24:
By specific value I mean 65636 bytes.

optical-o, 2022-11-24:
We are using it for small files and we did not experienced the issue with them.

When i tried to upload larger files such as nRF9160 Modem delta this occurred every time, however i'm not sure whether every attempt ended with the same error and at the same progress.

optical-o, 2022-11-24:
I can quickly check whether it happen at the same point with the same file.

optical-o, 2022-11-28:
I have been able to upload files larger than 64K to the device.
This PR #83  fixes the issue.

PR #83 (2022-11-28): Bug fix: support for files longer than 64k in FS manager
This PR fixes #78.
closed 2022-11-28 by philips77