nordicsemi/Android-nRF-Connect-Device-Manager #104: can't disconnect from device after firmware update

sjmerel, 2023-07-27:
I'm using the FirmwareUpgradeManager and it's working great, except at the end of the upgrade, it appears to still be connected to the device.  This means my own code can no longer connect to it as it's not advertising.

I start the process like this:
```
        this.transport = McuMgrBleTransport(activity, this.device.bluetoothDevice)
        val dfuManager = FirmwareUpgradeManager(this.transport!!, this.dfuCallback)

        val path = firmwarePath("firmware_$INSTALL_FIRMWARE_VERSION")
        val resId = activity.resources.getIdentifier(path, "raw", activity.packageName)
        val stream = activity.resources.openRawResource(resId)
        val buf = ByteArray(stream.available())
        stream.read(buf)

        dfuManager.setMode(FirmwareUpgradeManager.Mode.TEST_AND_CONFIRM)
        dfuManager.start(buf)
```

When FirmwareUpgradeCallback.onUpgradeCompleted() is called, I tried disconnecting explicitly:
```
        this.transport!!.disconnect()
```
but it didn't seem to have any effect.

Is there another way I should be disconnecting?

philips77, 2023-07-28:
Hello,
You're using `TEST_AND_CONFIRM` mode, in which the phone will upload the new image, send test command, end reset command, wait for disconnection, wait for swap to complete, reconnect and send confirm command. It will not disconnect afterwards.
Try using `CONFIRM_ONLY` mode, in which after uploading the new image a confirm command is sent immediately followed by the reset command. The library will not try to reconnect to the device afterwards.

sjmerelg, 2023-07-28:
Thanks! I also found that calling release() instead of disconnect() at the end would disconnect from the device.  Is this also a good solution?

closed 2024-01-05 by philips77