nordicsemi/Android-nRF-Connect-Device-Manager #110: [Request] Stop using features (such as 'sealed' in Kotlin) so as to keep the java-libs friendly towards the "Classic Xamarin" (mono) C# java-bindings toolchain

ksidirop-laerdal, 2023-09-01:
I've explained the issue we're facing here:

https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/issues/75#issuecomment-1702909223

In my team we created C# nugets that rely on java-bindings under the surface to interact with Nordic BLE devices. Up to version 1.5.1 of your libs everything was working smoothly. In the recent months however your libs started using 'sealed' classes:

https://github.com/search?q=repo%3ANordicSemiconductor%2FAndroid-nRF-Connect-Device-Manager%20sealed&type=code

When we upgraded the underlying java libs to version 1.7.2 we realized that even though our nuget packages where getting properly built (without any errors), when we tried to install+build "Classic Xamarin" apps for Android we would get a build-error like this one:

[image]

We tried to fork + refactor Nordic-Android-McuMgr ver 1.7.2 so that it won’t rely on the ‘sealed’ keyword but even when our wrapper was compiled against this modified version of the Nordic-Android-McuMgr library there was still an issue: The Sample App would compile and run with the new nugets but when we would try to perform a fw installation we would get a runtime error during the upload phase (could be related to the changes I made in those ‘sealed’ classes - not sure though - I'm still looking into this)

As far as I looked into this we either have to:

(a) keep the language features of your java libs in check with the "Classic Xamarin" toolchain (so no more 'sealed')

or (b) we will have to freeze support for "Classic Xamarin" and focus our aim to .Net6+ because only the updated r8 provided by the .Net6+ toolchain can cope with stuff like 'sealed' and so on - but even in this case if you keep using new java language features at will chances are that the C# toolchain will break again in random ways at some point thus forcing us to constantly support only the most bleeding-edge versions of .Net which is far from ideal in terms of adoption.

I will create a separate ticket asking whether its possible to stop using 'sealed' for starters.

ksidirop-laerdal, 2023-09-26:
I have created the following pull-request to resolve this issue:

https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/pull/114

philips77, 2023-10-22:
Merged. I hope it will work for you. I'm releasing 1.8.0 now.

closed 2023-10-22 by philips77

ksidirop-laerdal, 2023-10-24:
@philips77 Much appreciated. Will test things out asap and let you know how it panned out with the official .aars - our side will need this sort of compatibility just for the next 8-to-12 months or so. Xamarin will be completely phased out of the picture for us after this period. Again much appreciated and keep up the good work.

philips77, 2023-10-24:
I'll be releasing a bug-fixing version soon. If you find any issues, we may release another one any time.

ksidirop-laerdal, 2023-10-26:
> I'll be releasing a bug-fixing version soon. If you find any issues, we may release another one any time.

All our tests are green with 1.8.0 - again highly appreciated. Be safe.

philips77, 2023-10-27:
1.8.1 is greener.

ksidirop-laerdal, 2023-10-27:
> 1.8.1 is greener.

Appreciate bringing this on our radar - testing it out asap and keeping you posted.