nordicsemi/Android-nRF-Connect-Device-Manager #111: Detailed explanation of NO_ENTRY(5) error

olo95, 2023-09-04:
I'm receiving this error while trying to upload .bin file, could you provide more detailed explanation of this error? Can it be connected to .bin file itself?

"No entry was found for the request. This commonly means that the command group has not been enabled on the device, although the exact meaning."

philips77, 2023-09-12:
Hi, this is related to #112. In top right corner in the File System tab click the Settings icon and rename the partition to `/lfs1'. This apparently has changed 2 years ago in NCS.
More details in the mentioned issue.

philips77, 2023-09-12:
[image]

closed 2023-09-12 by philips77