nordicsemi/Android-nRF-Connect-Device-Manager #135: Minimum SDK issue with dependencies

alexkok, 2024-01-25:
Hello, we see a crash happening on Android 7 when using the `McuMgr`. 
Our app has a minimum SDK of 21, Android 5.

### Cause:

Tracing it down, the issue is caused by the Jackson dependency, see this corresponding issue: https://github.com/FasterXML/jackson-databind/issues/3658

As a result, the minimum SDK of Jackson got bumped to API 26 (Android 8) since version 2.14. Which resolves the issue for them.

Looking back at when McuMgr updated to Jackson 2.14+, it dates back to the commit at 22 feb 2023: https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/commit/006e5badfecb723c651437d926b40568e2d88878

### Issue/question:

Since McuMgr Android currently describes the requirement of Android 5 as minimum.
Would this mean that the minimum SDK for McuMgr Android should become Android 8? 
Or will there come another solution for this?

### Workaround:

As a workaround we can use the old McuMgr (version 1.5.2). That version depends on Jackson version 2.13.3 which doesn't have this issue. But we would lack the fixes and features of the later versions of McuMgr of course.

### Stacktrace:

For completeness, here is the error stacktrace (happens since version 1.6.0 and the latest version 1.9.1):

```
java.lang.NoSuchMethodError: No virtual method getParameterCount()I in class Ljava/lang/reflect/Constructor; or its super classes (declaration of 'java.lang.reflect.Constructor' appears in /system/framework/core-oj.jar)
    at com.fasterxml.jackson.databind.util.ClassUtil$Ctor.getParamCount(ClassUtil.java:1450)
    at com.fasterxml.jackson.databind.introspect.AnnotatedCreatorCollector._findPotentialConstructors(AnnotatedCreatorCollector.java:120)
    at com.fasterxml.jackson.databind.introspect.AnnotatedCreatorCollector.collect(AnnotatedCreatorCollector.java:70)
    at com.fasterxml.jackson.databind.introspect.AnnotatedCreatorCollector.collectCreators(AnnotatedCreatorCollector.java:61)
    at com.fasterxml.jackson.databind.introspect.AnnotatedClass._creators(AnnotatedClass.java:403)
    at com.fasterxml.jackson.databind.introspect.AnnotatedClass.getConstructors(AnnotatedClass.java:308)
    at com.fasterxml.jackson.databind.introspect.POJOPropertiesCollector._addCreators(POJOPropertiesCollector.java:618)
    at com.fasterxml.jackson.databind.introspect.POJOPropertiesCollector.collectAll(POJOPropertiesCollector.java:431)
    at com.fasterxml.jackson.databind.introspect.POJOPropertiesCollector.getPropertyMap(POJOPropertiesCollector.java:391)
    at com.fasterxml.jackson.databind.introspect.POJOPropertiesCollector.getProperties(POJOPropertiesCollector.java:233)
    at com.fasterxml.jackson.databind.introspect.BasicBeanDescription._properties(BasicBeanDescription.java:164)
    at com.fasterxml.jackson.databind.introspect.BasicBeanDescription.findProperties(BasicBeanDescription.java:239)
    at com.fasterxml.jackson.databind.deser.BasicDeserializerFactory._findCreatorsFromProperties(BasicDeserializerFactory.java:328)
    at com.fasterxml.jackson.databind.deser.BasicDeserializerFactory._constructDefaultValueInstantiator(BasicDeserializerFactory.java:272)
    at com.fasterxml.jackson.databind.deser.BasicDeserializerFactory.findValueInstantiator(BasicDeserializerFactory.java:223)
    at com.fasterxml.jackson.databind.deser.BeanDeserializerFactory.buildBeanDeserializer(BeanDeserializerFactory.java:262)
    at com.fasterxml.jackson.databind.deser.BeanDeserializerFactory.createBeanDeserializer(BeanDeserializerFactory.java:151)
    at com.fasterxml.jackson.databind.deser.DeserializerCache._createDeserializer2(DeserializerCache.java:415)
    at com.fasterxml.jackson.databind.deser.DeserializerCache._createDeserializer(DeserializerCache.java:350)
    at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCache2(DeserializerCache.java:264)
    at com.fasterxml.jackson.databind.deser.DeserializerCache._createAndCacheValueDeserializer(DeserializerCache.java:244)
    at com.fasterxml.jackson.databind.deser.DeserializerCache.findValueDeserializer(DeserializerCache.java:142)
    at com.fasterxml.jackson.databind.DeserializationContext.findRootValueDeserializer(DeserializationContext.java:648)
    at com.fasterxml.jackson.databind.ObjectMapper._findRootDeserializer(ObjectMapper.java:4861)
    at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:4731)
    at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:3738)
    at io.runtime.mcumgr.util.CBOR.toObject(CBOR.java:31)
    at io.runtime.mcumgr.response.McuMgrResponse.buildResponse(McuMgrResponse.java:247)
    at io.runtime.mcumgr.ble.McuMgrBleTransport$2.onResponse(McuMgrBleTransport.java:409)
    at io.runtime.mcumgr.ble.callback.SmpProtocolSessionKt.onResponse$lambda$1(SmpProtocolSession.kt:156)
    at io.runtime.mcumgr.ble.callback.SmpProtocolSessionKt.$r8$lambda$y5OWzRv1IbLlvC7QCOqni68WVyI(SmpProtocolSession.kt)
    at io.runtime.mcumgr.ble.callback.SmpProtocolSessionKt$$ExternalSyntheticLambda2.run(D8$$SyntheticClass)
    at android.os.Handler.handleCallback(Handler.java:751)
    at android.os.Handler.dispatchMessage(Handler.java:95)
    at android.os.Looper.loop(Looper.java:154)
    at android.app.ActivityThread.main(ActivityThread.java:6780)
    at java.lang.reflect.Method.invoke(Native Method)
    at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1500)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1390)
```

philips77, 2024-01-25:
I'll try to revert to 2.13.3. Thank you for the information.

closed 2024-01-25 by philips77

philips77, 2024-01-25:
Released as 1.9.2.

PR #136 (2024-01-25): Reverting jackson lib to 2.13.5
This PR fixes #135.

It reverts Jackson CBOR library to 2.13.5, latest version not requiring Android 8. 

> [!NOTE]
> Seems like despite requirement of Android 7 for version 2.13.5, the features used in this library work on older Android versions.