nordicsemi/Android-nRF-Connect-Device-Manager #138: Reading multiple Settings via settingsManager returns same result for different settings

akrenz, 2024-01-27:
I am trying to read multiple settings from inside a composable from my Device via the `settingsManager`. I am using the following code

```
@Composable
    fun DetailsScreen(navController: NavController) {
        val device = bluetoothViewModel.selectedDevice ?: return
        val context = LocalContext.current

        fun readSetting(device: BluetoothDevice, settingName: String, settingState: MutableState<String?>) {
            val transport = McuMgrBleTransport(context, device)
            val settingsManager = SettingsManager(transport)
            Log.d("readSetting","Start reading Setting ${settingName}")

            settingsManager.read(
                settingName,
                object : McuMgrCallback<McuMgrSettingsReadResponse> {
                    override fun onResponse(response: McuMgrSettingsReadResponse) {
                        if (response.isSuccess) {
                            val byteArray = response.`val`
                            settingState.value = byteArray.toString(Charsets.UTF_8)
                            Log.d("Callback", "onResponse called. Setting: $settingName, Response: ${settingState.value}")
                        }
                    }

                    override fun onError(p0: McuMgrException) {
                        Toast.makeText(
                            context,
                            "Konnte Setting \"$settingName\" nicht lesen: ${p0.message}, ${p0.cause}",
                            Toast.LENGTH_LONG
                        ).show()
                    }
                })
        }

        val ssidState = remember { mutableStateOf<String?>(null) }
        val passwordState = remember { mutableStateOf<String?>(null) }

        // Rufen Sie die Funktion für die gewünschten Einstellungen auf
        LaunchedEffect(device){
                readSetting(device, "wifi/ssid", ssidState)
                readSetting(device, "wifi/password", passwordState)
        }

        Surface(modifier = Modifier.fillMaxSize(), color = MaterialTheme.colorScheme.background) {
            Column {
                if (device != null) {
                    Column(
                        modifier = Modifier.weight(1f),
                        verticalArrangement = Arrangement.Top
                    ) {
                        Text(text = "${device.name}")
                        Text(text = "Wifi SSID: ${ssidState.value ?: "Lade..."}")
                        Text(text = "Wifi Password: ${passwordState.value ?: "Lade..."}")
                    }
                } else {
                    Text(text = "Kein Gerät ausgewählt")
                }

                // Button zum Zurückkehren zur vorherigen Ansicht
                Button(onClick = { navController.popBackStack() }) {
                    Text("Zurück")
                }
            }
        }
    }
```
The android logs show the same string result for both requests:
```
2024-01-27 01:09:18.546  6535-6535  Callback                com.example.test                     D  onResponse called. Setting: wifi/ssid, Response: <my-ssid-string>??
2024-01-27 01:09:18.564  6535-6535  Callback                com.example.test                     D  onResponse called. Setting: wifi/password, Response: <my-ssid-string>??
```
I can see that my device returns the right content (printed in the logs):

```
[00:48:43.819,702] <inf> settings_mgmt: Read Key: wifi/ssid
[00:48:43.819,732] <inf> app_settings: Look for Key: ssid, max_length: 32
[00:48:43.819,793] <inf> app_settings: Found content: <my-ssid-string>, length: 9
[00:48:43.907,989] <inf> settings_mgmt: Read Key: wifi/password
[00:48:43.908,020] <inf> app_settings: Look for Key: password, max_length: 32
[00:48:43.908,050] <inf> app_settings: Found content: <here-is-my-password>, length: 12
```

Unfortunately both texts in my `Surface` show the same string (either SSID or password). How do I read multiple settings in parallel?

akrenz, 2024-02-03:
I was able to solve this by wrapping all `readSetting` in a `suspendCoroutine` and make them blocking calls:

```
suspend fun readSettingString(device: BluetoothDevice, settingName: String): String? =
            suspendCoroutine {cont ->
                val transport = McuMgrBleTransport(context, device)
                val settingsManager = SettingsManager(transport)

                val callback = object : McuMgrCallback<McuMgrSettingsReadResponse> {
                    override fun onResponse(response: McuMgrSettingsReadResponse) {
                        if (response.isSuccess) {
                            val byteArray = response.`val`
                            cont.resume(byteArray.decodeToString(0, byteArray.indexOf(0.toByte())))
                        }
                    }

                    override fun onError(p0: McuMgrException) {

                        cont.resume(null)
                    }
                }

                // Read setting asynchronously
                settingsManager.read(settingName, callback)
            }
```

Now I can read one setting after another inside a `LaunchedEffect` and write the result to my `MutableState<String>` elements

```
LaunchedEffect(device){
            ssidState.value = readSettingString(device, "wifi/ssid")
            passwordState.value = readSettingString(device, "wifi/password")
            serverState.value = readSettingString(device, "lwm2m/server")
        }
```

philips77, 2024-02-06:
THank you ofr sharing. Out of curiosity, in your first approach, did you enable logs from `McuMgrTransport`?
You may enable them using the following methods:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/6411d43d547018a722cbd23e4f6bb28b21ac46be/mcumgr-ble/src/main/java/io/runtime/mcumgr/ble/McuMgrBleTransport.java#L286-L298
Or just always return `Log.VERBOSE` from the second one.

I wonder where the Strings get overriden. Does it happen on the BLE layer, or above.

akrenz, 2024-02-09:
I tried with enabling all logs via `transport.setLoggingEnabled(true)`. But I don't get much more logs out of it:

```
024-02-09 21:44:02.241 14035-14035 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/device_type
2024-02-09 21:44:02.245 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 21:44:02.245 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 21:44:02.245 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=311726aa-d1d6-4bd8-98c7-af26fab5fe4d
2024-02-09 21:44:02.247 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=9
2024-02-09 21:44:02.248 14035-14035 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/manufacturer
2024-02-09 21:44:02.249 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=9 device=F2:ED:A3:C8:1B:7F
2024-02-09 21:44:02.250 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 21:44:02.250 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 21:44:02.250 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 21:44:02.250 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=603a0d74-5e69-4a64-acf3-c59dffd7d974
2024-02-09 21:44:02.251 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=10
2024-02-09 21:44:02.252 14035-14035 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/hardware_version
2024-02-09 21:44:02.254 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 21:44:02.254 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 21:44:02.254 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=58016260-ae8b-478f-a6b4-62a88aeb8adb
2024-02-09 21:44:02.254 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=10 device=F2:ED:A3:C8:1B:7F
2024-02-09 21:44:02.254 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 21:44:02.255 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=11
2024-02-09 21:44:02.256 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=11 device=F2:ED:A3:C8:1B:7F
2024-02-09 21:44:02.256 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 21:44:02.603 14035-14149 BluetoothGatt           com.example.oha_provisioning         D  discoverServices() - device: XX:XX:XX:XX:1B:7F
2024-02-09 21:44:02.603 14035-14151 BluetoothGatt           com.example.oha_provisioning         D  discoverServices() - device: XX:XX:XX:XX:1B:7F
2024-02-09 21:44:02.604 14035-14150 BluetoothGatt           com.example.oha_provisioning         D  discoverServices() - device: XX:XX:XX:XX:1B:7F
2024-02-09 21:44:02.610 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onSearchComplete() = Device=F2:ED:A3:C8:1B:7F Status=0
2024-02-09 21:44:02.612 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  setCharacteristicNotification() - uuid: da2e7828-fbce-4e01-ae9e-261174997c48 enable: true
2024-02-09 21:44:02.614 14035-14051 BluetoothGatt           com.example.oha_provisioning         D  onSearchComplete() = Device=F2:ED:A3:C8:1B:7F Status=0
2024-02-09 21:44:02.615 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onSearchComplete() = Device=F2:ED:A3:C8:1B:7F Status=0
2024-02-09 21:44:02.621 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  setCharacteristicNotification() - uuid: da2e7828-fbce-4e01-ae9e-261174997c48 enable: true
2024-02-09 21:44:02.628 14035-14035 BluetoothGatt           com.example.oha_provisioning         D  setCharacteristicNotification() - uuid: da2e7828-fbce-4e01-ae9e-261174997c48 enable: true
2024-02-09 21:44:03.188 14035-14035 Callback                com.example.oha_provisioning         D  onResponse called. Setting: lwm2m/device_type, Response: S0-Logger??
2024-02-09 21:44:03.205 14035-14035 Callback                com.example.oha_provisioning         D  onResponse called. Setting: lwm2m/manufacturer, Response: S0-Logger??
2024-02-09 21:44:03.218 14035-14035 Callback                com.example.oha_provisioning         D  onResponse called. Setting: lwm2m/hardware_version, Response: S0-Logger??
2024-02-09 21:44:03.294 14035-14055 BluetoothGatt           com.example.oha_provisioning         D  onConnectionUpdated() - Device=F2:ED:A3:C8:1B:7F interval=9 latency=0 timeout=42 status=0
[... cut ...]
2024-02-09 21:44:08.283 14035-14052 BluetoothGatt           com.example.oha_provisioning         D  onConnectionUpdated() - Device=F2:ED:A3:C8:1B:7F interval=36 latency=0 timeout=42 status=0
```

I tried adding this library as sourcecode to my project. but i failed. I do not know anything about Android Developement so I dont know how to add the library (tried with some googling but failed because of dependencies between `mcumgr-core` and `mcumgr-ble`)

Maybe library has a problem because I am creating an `McuMgrBleTransport`  and `SettingsManager` instance of `readSettings`?

akrenz, 2024-02-09:
After changing the code to use logcat I got the following log output:

```
2024-02-09 23:05:42.317 20891-20891 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/device_type
2024-02-09 23:05:42.319 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         V  Connecting...
2024-02-09 23:05:42.319 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-02-09 23:05:42.320 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 23:05:42.320 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 23:05:42.320 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=5d440741-ccd3-4593-aaca-83ed27dc3ded
2024-02-09 23:05:42.323 20891-20952 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=9
2024-02-09 23:05:42.323 20891-20891 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/manufacturer
2024-02-09 23:05:42.325 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         V  Connecting...
2024-02-09 23:05:42.325 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-02-09 23:05:42.325 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=9 device=F2:ED:A3:C8:1B:7F
2024-02-09 23:05:42.325 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 23:05:42.325 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 23:05:42.325 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=de0799dd-115c-4833-a166-cb0040a638c2
2024-02-09 23:05:42.326 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 23:05:42.326 20891-20952 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=10
2024-02-09 23:05:42.326 20891-20891 readSetting             com.example.oha_provisioning         D  Start reading Setting lwm2m/hardware_version
2024-02-09 23:05:42.328 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         V  Connecting...
2024-02-09 23:05:42.328 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-02-09 23:05:42.329 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  connect() - device: XX:XX:XX:XX:1B:7F, auto: false
2024-02-09 23:05:42.329 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp()
2024-02-09 23:05:42.329 20891-20891 BluetoothGatt           com.example.oha_provisioning         D  registerApp() - UUID=42eec4dd-8b30-4fed-9007-e3bbbbd5a21b
2024-02-09 23:05:42.330 20891-20952 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=10 device=F2:ED:A3:C8:1B:7F
2024-02-09 23:05:42.330 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 23:05:42.332 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onClientRegistered() - status=0 clientIf=11
2024-02-09 23:05:42.336 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onClientConnectionState() - status=0 clientIf=11 device=F2:ED:A3:C8:1B:7F
2024-02-09 23:05:42.336 20891-20912 BluetoothGatt           com.example.oha_provisioning         D  onConfigureMTU() - Device=F2:ED:A3:C8:1B:7F mtu=498 status=0
2024-02-09 23:05:42.367 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2024-02-09 23:05:42.367 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connected to F2:ED:A3:C8:1B:7F
2024-02-09 23:05:42.367 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  wait(300)
2024-02-09 23:05:42.367 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  MTU changed to: 498
2024-02-09 23:05:42.367 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2024-02-09 23:05:42.368 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connected to F2:ED:A3:C8:1B:7F
2024-02-09 23:05:42.368 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  wait(300)
2024-02-09 23:05:42.368 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  MTU changed to: 498
2024-02-09 23:05:42.368 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         D  [Callback] Connection state changed with status: 0 and new state: 2 (CONNECTED)
2024-02-09 23:05:42.368 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connected to F2:ED:A3:C8:1B:7F
[... cut ...]
2024-02-09 23:05:43.391 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
2024-02-09 23:05:43.391 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
2024-02-09 23:05:43.391 20891-20891 McuMgrBleTransport      com.example.oha_provisioning         I  Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
```

akrenz, 2024-02-09:
After changing my code to create only a single instance of `SettingsMananger` the responses are now handled correctly:

```
fun DetailsScreen(navController: NavController, bluetoothViewModel: BluetoothViewModel) {
    val device = bluetoothViewModel.selectedDevice ?: return
    val context = LocalContext.current
    var devName = remember { mutableStateOf<String?>(null) }
    var devManufacturer = remember { mutableStateOf<String?>(null) }
    var devHardwareVersion = remember { mutableStateOf<String?>(null) }
    var ssidState = remember { mutableStateOf<String?>(null) }
    var passwordState = remember { mutableStateOf<String?>(null) }
    var serverState = remember { mutableStateOf<String?>(null) }
    var techState = remember { mutableStateOf<UInt>(0u) }

    val transport = McuMgrBleTransport(context, device)
    val settingsManager = SettingsManager(transport)

// ... some other code

fun SettingsManager.readSetting(settingName: String, state: MutableState<String?>) {
        Log.d("readSetting","Start reading Setting ${settingName}")

        read(
            settingName,
            object : McuMgrCallback<McuMgrSettingsReadResponse> {
                override fun onResponse(response: McuMgrSettingsReadResponse) {
                    if (response.isSuccess) {
                        val byteArray = response.`val`
                        state.value = byteArray.toString(Charsets.UTF_8)
                        Log.d("readSetting", "onResponse called. Setting: $settingName, Response: ${state.value}")
                    }
                }

                override fun onError(p0: McuMgrException) {
                    Log.e("readSetting", "Could not read setting $settingName. Error: ${p0.message}")
                    Toast.makeText(
                        context,
                        "Konnte Setting \"$settingName\" nicht lesen: ${p0.message}, ${p0.cause}",
                        Toast.LENGTH_LONG
                    ).show()
                }
            })
    }

    fun SettingsManager.readSetting(settingName: String, settingState: MutableState<UInt>) {
            val callback = object : McuMgrCallback<McuMgrSettingsReadResponse> {
                override fun onResponse(response: McuMgrSettingsReadResponse) {
                    if (response.isSuccess) {
                        val byteArray = response.`val`
                        settingState.value = byteArray.toUInt()
                        Log.d("readSetting", "onResponse called. Setting: $settingName, Response: ${byteArray.toHex()}")
                    }
                }

                override fun onError(p0: McuMgrException) {
                    Log.e("readSetting", "Could not read setting $settingName. Error: ${p0.message}")
                    Toast.makeText(
                        context,
                        "Konnte Setting \"$settingName\" nicht lesen: ${p0.message}, ${p0.cause}",
                        Toast.LENGTH_LONG
                    ).show()
                }
            }

            // Read setting asynchronously
            read(settingName, callback)
        }

// ... some other code here

    LaunchedEffect(device){
        settingsManager.readSetting("lwm2m/device_type", devName)
        settingsManager.readSetting("lwm2m/manufacturer", devManufacturer)
        settingsManager.readSetting("lwm2m/hardware_version", devHardwareVersion)
        settingsManager.readSetting("wifi/ssid", ssidState)
        settingsManager.readSetting("wifi/password", passwordState)
        settingsManager.readSetting("lwm2m/server", serverState)
        settingsManager.readSetting("tech/imp_per_kwh", techState)
    }
}
```

It should be documented somewhere that you must not create multiple instances of `SettingsMananger`. Or better: use some kind of Singleton pattern.

akrenz, 2024-02-24:
Any update on this?

philips77, 2024-02-26:
Not yet, sorry. We're busy with other tasks.