nordicsemi/Android-nRF-Connect-Device-Manager #139: Migration to Kotlin BLE library

albertmoravec, 2024-01-29:
Hi, I just wanted to ask if there are any plans regarding migration to Kotlin BLE library. As I understand it, as long as I am using managers from this library, I am locked to using `McuMgrBleTransport` and the old Android BLE library, is that correct?

Of course I could use both libraries at the same time, but that kind of defeats the purpose of migrating and makes things a lot more complicated.

Looking forward to hear about next steps.

philips77, 2024-02-05:
Hi,
That is a good point, I understand your concerns. However, currently we have other priorities, and the Kotlin BLE Library isn't mature enough. Eventually we will migrate, but I can't promise any deadline.

I'm leaving this issue open as an open task.

LightWeek, 2024-02-07:
@albertmoravec Hi. Have you found any reasonable solution to use this library along with Kotlin BLE library ?

LightWeek, 2024-02-07:
@philips77 Of course. That should have been pretty obvious. I should have looked at the code first before asking questions. Anyways, thanks. Implementing custom McuMgrBleTransport wrapper did the job.