nordicsemi/Android-nRF-Connect-Device-Manager #142: Upgrade mode "NONE" with no reset

mateusz-, 2024-02-27:
On the firmware side, MCUmgr supports DFU callbacks. Specifically, there's an event called `MGMT_EVT_OP_IMG_MGMT_DFU_PENDING` which occurs when an image finished being transferred.

I haven't used this library yet, but for the "NONE" firmware upgrade mode, the README.md states that:
> The process for this mode is `UPLOAD`, `RESET`.

What if I want to use `MGMT_EVT_OP_IMG_MGMT_DFU_PENDING` to perform some post-upload work and handle test / confirm on the firmware side? I would not want to reset right after upload.

Should there be a `NONE_NO_RESET` upgrade mode?

philips77, 2024-02-28:
You may still use the `ImageManager` directly, like on the Advanced screen of the nRF Connect Device Manager app.
The `FirmwareUpdateManager` is using this and `DefaultManager` (OS Manager) for atomic operations.

mateusz-, 2024-02-29:
Okay, thank you!

closed 2024-02-29 by mateusz-