nordicsemi/Android-nRF-Connect-Device-Manager #146: Question: Timeout while updating firmware

archie94, 2024-04-08:
On certain device when we try to update device firmware we get a timeout with the following

```
ERROR msg = Transaction 67 timed out without receiving a response
```

Generally a bluetooth toggle and retry fixes it. 

What can be the issue here?

archie94, 2024-04-10:
@philips77 do you have any idea on this one?

philips77, 2024-09-06:
Hi,
Sorry for the late response. I must have missed the question.
It may be, that some recent changes improved the situation. Did you try versions 2.0.x?
Also, you may decrease number of buffers used using smaller _windows capacity_. Setting it to 1 will make the library always await notification before sending a new packet, making upload slower, but this issue should never happen. But you may try with value 2 or 3.

closed 2024-09-06 by philips77