nordicsemi/Android-nRF-Connect-Device-Manager #147: File download command

giorgiog73, 2024-04-29:
I use the device manager to test my PCB (BLE) (nordic 52845)running Zephyr (NCS 2.6). I can send hellos, perform DFUs, and write/read files. I was looking for the command that the app uses to read a file because I am looking for a way to read the file and save it to the Android mobile phone. Is the file download command use a mcumgr command? Is it possible to replicate this command and send it as shell command?
Thanks for your help
Giorgio

philips77, 2024-09-06:
You may use the `FileManager` to download a file. It will give you a byte array when done. It's using `download` command from the File group on mcu mgr. 

> Is it possible to replicate this command and send it as shell command?

You may create your own manager and implement commands as you wish.

philips77, 2024-09-06:
And sorry for the late response! I must have missed your question.

closed 2024-09-06 by philips77