nordicsemi/Android-nRF-Connect-Device-Manager #156: Issue with working with the McuMgr together with the Android BLE library

talfsaverone, 2024-05-30:
Hello,

We are working with the Nordic SDK for 2 purposes that are running together.

1. We connect our Bluetooth peripheral with the Android BLE library to send and receive messages.
2. While connected with the Android BLE library we need to upload and download files with the McuMgrBleTransport 

For some reason, we get timeouts while uploading/downloading large files and sudden disconnections while connected to the Android BLE library and the McuMgrBleTransport together.

What can we do in this situation?

philips77, 2024-05-31:
I don't think this is related to any of the mobile libraries. I'd say it may be something with timings, e.g. when the device is quickly saving incoming data it may slow down responses every now and then, causing the phone to drop connection. Did you experiment with using different connection parameters? You may request them from the Android or peripheral side. Mind, that longer connection interval does not necessarily mean slower upload, as sometimes more _write without response_ packets can fit into longer connection interval. You may also experiment with using lower window capacity (less packets sent in parallel).

Do your logs show anything intersting apart from the error code?

talfsaverone, 2024-06-02:
@philips77 

I see interesting things in the Logs, upon disconnecting our Bluetooth peripheral device the McuMgrBleTransport is trying to auto-connect to the device for some reason (maybe you can further explain this):
`2024-06-02 15:17:44.794 McuMgrBleTransport   com.saverone.app                     D  Writing characteristic da2e7828-fbce-4e01-ae9e-261174997c48 (WRITE COMMAND)
2024-06-02 15:17:44.794 McuMgrBleTransport   com.saverone.app                     D  gatt.writeCharacteristic(da2e7828-fbce-4e01-ae9e-261174997c48, value=0x323DDA649CC87A17CEA821B5A3EF0CA9F528956E5B3221C9DCB0D4B35FE7DE915CAA2D9E0A99C336BA508C8E4DE91E14C7AC0BF9067711F46BE9AB2C7DB3B2A59ED3B4AD48FB16A39DF2FE57387DECBAB3DD5487410C871BD2039B79172B7D9FAC3DDDF11CE172594F7DF5343AFB669A5ED539487D13282C97AF99B96A0376888B8F16F3869BD5F5EBF75ED5DABC38ACEBBB53E9079C1C58B724F419685CD865CE9E744F7D4D9D736ED5708220BD47F6B597CC1ED2B786F445D3E2636F66661A000BBD90FF, WRITE COMMAND)
2024-06-02 15:17:44.799 McuMgrBleTransport   com.saverone.app                     D  Data written to da2e7828-fbce-4e01-ae9e-261174997c48
2024-06-02 15:17:49.996 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2024-06-02 15:17:49.996 McuMgrBleTransport   com.saverone.app                     D  Disconnected
2024-06-02 15:17:50.019 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:17:50.034 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 0 and new state: 0 (DISCONNECTED)
2024-06-02 15:17:50.034 McuMgrBleTransport   com.saverone.app                     D  Disconnected
2024-06-02 15:17:50.035 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:17:50.103 McuMgrBleTransport   com.saverone.app                     D  Connecting...
2024-06-02 15:17:50.103 McuMgrBleTransport   com.saverone.app                     D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-06-02 15:17:55.165 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2024-06-02 15:17:55.165 McuMgrBleTransport   com.saverone.app                     D  Error: (0x85): GATT ERROR
2024-06-02 15:17:55.165 McuMgrBleTransport   com.saverone.app                     D  wait(500)
2024-06-02 15:17:55.669 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:17:55.700 McuMgrBleTransport   com.saverone.app                     D  wait(200)
2024-06-02 15:17:55.901 McuMgrBleTransport   com.saverone.app                     D  Retrying...
2024-06-02 15:17:55.901 McuMgrBleTransport   com.saverone.app                     D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-06-02 15:18:00.992 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2024-06-02 15:18:00.992 McuMgrBleTransport   com.saverone.app                     D  Error: (0x85): GATT ERROR
2024-06-02 15:18:00.992 McuMgrBleTransport   com.saverone.app                     D  wait(500)
2024-06-02 15:18:01.496 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:18:01.535 McuMgrBleTransport   com.saverone.app                     D  wait(200)
2024-06-02 15:18:01.738 McuMgrBleTransport   com.saverone.app                     D  Retrying...
2024-06-02 15:18:01.739 McuMgrBleTransport   com.saverone.app                     D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-06-02 15:18:06.813 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2024-06-02 15:18:06.813 McuMgrBleTransport   com.saverone.app                     D  Error: (0x85): GATT ERROR
2024-06-02 15:18:06.813 McuMgrBleTransport   com.saverone.app                     D  wait(500)
2024-06-02 15:18:07.315 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:18:07.341 McuMgrBleTransport   com.saverone.app                     D  wait(200)
2024-06-02 15:18:07.544 McuMgrBleTransport   com.saverone.app                     D  Retrying...
2024-06-02 15:18:07.544 McuMgrBleTransport   com.saverone.app                     D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
2024-06-02 15:18:12.451 McuMgrBleTransport   com.saverone.app                     D  [Callback] Connection state changed with status: 133 and new state: 0 (DISCONNECTED)
2024-06-02 15:18:12.452 McuMgrBleTransport   com.saverone.app                     D  Error: (0x85): GATT ERROR
2024-06-02 15:18:12.452 McuMgrBleTransport   com.saverone.app                     D  wait(500)
2024-06-02 15:18:12.956 McuMgrBleTransport   com.saverone.app                     D  gatt.close()
2024-06-02 15:18:12.994 McuMgrBleTransport   com.saverone.app                     D  wait(200)
2024-06-02 15:18:13.195 McuMgrBleTransport   com.saverone.app                     D  Retrying...
2024-06-02 15:18:13.196 McuMgrBleTransport   com.saverone.app                     D  gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
[... cut ...] 330302E352E737775FF, WRITE COMMAND)
2024-06-02 15:18:50.596 McuMgrBleTransport   com.saverone.app                     D  Data written to da2e7828-fbce-4e01-ae9e-261174997c48
2024-06-02 15:18:52.716 McuMgrBleTransport   com.saverone.app                     D  Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
2024-06-02 15:18:52.719 McuMgrBleTransport   com.saverone.app                     D  Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)`

philips77, 2024-06-04:
The first long packet after reconnection that you send, the one of size 2474 bytes, has `off = 767012`:
> Header (Version: 1, Op: 2, Flags: 0, Len: 2466, Group: 8, Seq: 0, Command: 0) 
CBOR {"name":"/data/swu/pluv2_v1.300.5.swu","data":"H5a7sdyDZ...DTXvz","off":767012}

Seems like the app is trying to continue sending the packet, but the device might have lost the context on reboot, so may be expecting `off = 0`, perhaps that's why it isn't responding for 2 seconds. Or it needs somewhat more time to reply.

talfsaverone, 2024-06-05:
I understand, thanks for your response.

Our use case involves making several file uploads/downloads to our Bluetooth device one by one while passing other messages on different services.

So what do you think is the best way to handle the `FsManager ` in that case?

Create a new `FsManager` for every request (and clear it in some way?), or use the same `FsManager` for all requests?

Also, what is the best way to gracefully clear and close the resources of the `FsManager` and the `McuMgrBleTransport` upon disconnection?

talfsaverone, 2024-06-05:
I think I know what is the issue of the connecting after disconnection.

Every send call is calling connect over and over again with a retry of 5 and 500 ms delay.

Therefore if the send is called just before disconnection we are entering this loop causing the above issue.

philips77, 2024-09-06:
The BLE transport implementation indeed initiates connection when not already connected.
You may check the connection state before sending the new value.