nordicsemi/Android-nRF-Connect-Device-Manager #173: 2.0.0 Crashes as soon as you tap on a device

jgilles, 2024-08-08:
Using Thing53 with Nordic sample code and the Device Manger app.  Device shows up in in the list of devices and as soon as you tap on it, the app crashes.  Version 1.9.2 worked without problems.

daehyeonyim, 2024-08-09:
I also met the same problem even if using SMP server sample with the nRF52840 DK.

dastarling, 2024-08-11:
We are also seeing this issue as well with custom device that was working fine in previous version.

zwoabier, 2024-08-12:
Same for me

hhpettersen, 2024-08-12:
Forking repo excluding commits after 05/07 works: https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/tree/76817fff22924f04d2e4c3ea58cb97d01dd4d714

philips77, 2024-08-16:
Hello, 
Thank you for the report. I'll fix it immediately. Please expect new version in few hours.

philips77, 2024-08-16:
I just released version 2.0.1. Should be available in the Play Store shortly. Once again sorry for that and thank you for details and the stack trace.

PR #174 (2024-08-16): Fixed crash when nRF Logger is not installed
This PR fixes #173.
closed 2024-08-16 by philips77