nordicsemi/Android-nRF-Connect-Device-Manager #188: Question: Are the concepts of windowCapacity and memoryAlignment applicable to file-downloading or are they only applicable to file-uploading

ksidirop-laerdal, 2024-10-01:
I noticed that there's an asymmetry between the way we initiate a file-uploading vs file-downloading. To illustrate my point:

- In file-uploading we have:

```java
var _uploadController = new FileUploader( //00
                    _fileSystemManager,
                    _remoteFilePathSanitized,
                    data,
                    windowCapacity,     <-----
                    memoryAlignment   <-----
).uploadAsync(_fileUploaderCallbackProxy);
```

- In file-downloading we have:

```java
_downloadingController = _fileSystemManager.fileDownload(remoteFilePath, new FileDownloaderCallbackProxy());
```

I can't see any way to specify the window-capacity or the memory-alignment parameters to the file-downloader. Am I missing something obvious or is this to be expected because these notions don't apply to file-downloads?

Appreciate any insights.

philips77, 2024-10-03:
The `memoryAlignment` parameter is indeed valid only for the transmitter (so only when uploading, or on the device side when downloading to a phone), but we could think of this window capacity. I don't even know it that's supported by the firmware in the current SDKs... I think it should be, never tested.

ksidirop-laerdal, 2024-10-22:
I'm sated - you may close this one to keep the list clean ;) @philips77

closed 2024-10-23 by philips77