nordicsemi/Android-nRF-Connect-Device-Manager #189: Is it possible to support UART McuMgrTransport instead of BLE

Motirn, 2024-10-07:
Any foreseeable issues in implementing a uart based transport instead of BLE?
I have a requirement to update firmware of a nrf52840 device connected to andorid device over serial port. And has restrictions on using bluetooth for updating firmware.

Any issues to be expected if putting together a 'serial' McuMgrTransport.
Any help or direction is much appreciated.
Thanks

philips77, 2024-10-08:
Hello,
Serial implementation is available on Zephyr project (in C, GO, or something like that).
We do not plan to add serial transport to this library, but the library is written in transport-agnostic way, so you could try making your own implementation if needed.

philips77, 2024-10-08:
You may find some information here: https://docs.zephyrproject.org/latest/services/device_mgmt/smp_transport.html#uart-serial-and-console

closed 2024-10-08 by philips77