nordicsemi/Android-nRF-Connect-Device-Manager #200: File-Downloader fails start on Samsung A8 that connects in Phy1M mode to nRF51+ chipset

ksidirop-laerdal, 2024-10-24:
Even though we had success making Samsung A8 work in terms of uploading files/firmwares to our devices via:

Phy1M mode + initial-mtu-size=23 + window-size=1

We can't repeat this on the file-downloader using:

Phy1M mode + initial-mtu-size=<23 or any value all the way up to 100> (window-size is not supported in the file-downloader)

We get an error instead:

'Transaction 0 timed out without receiving a response'

The file-downloader works great using the same code on any other Android device even in Phy1M if we use initial-mtu-size >= 30. It's only Samsung A8 that suffers from this shortcoming.

( We suspect it might have to do with the fact that we can't set the window-size like we do with the file-uploader - but that's just a blind guess )

Any ideas as to what we might be doing wrong are welcome. Code sample:

```
        _transport = new McuMgrBleTransport(_context, _bluetoothDevice);
        if (initialMtuSize > 0)
        {
            _transport.setInitialMtu(initialMtuSize);
        }

        _transport.requestConnPriority(ConnectionPriorityRequest.CONNECTION_PRIORITY_HIGH);

       _fileSystemManager = new FsManager(_transport); //order

       _downloadingController = _fileSystemManager.fileDownload(_remoteFilePathSanitized, new FileDownloaderCallbackProxy());
```

LightWeek, 2024-11-22:
I am encountering the same error on a Pixel 3a running Android 12 when attempting to download a file from our nRF52 device. The error consistently reproduces in the **nRF Device Manager** app and in our app using McuMgrBleTransport + FsManager (code as simple as in the original post above).

[image]

However, downloading the same file from the same nRF52 device works as expected on other test phones (ranging from Android 12 to Android 15 and iOS).

LightWeek, 2024-11-27:
@ksidirop-laerdal Have you found any solution to this issue?
@philips77 Do you know possible cause or solution? Should I provide any more information or logs? 

BTW. For some reason logger app is not available on Google Play and one installed from APK doesn't connect with Device Manager app.
[image]

philips77, 2024-11-27:
Oh, great catch. I need to update the target SDK in the app to make it available again.

ksidirop-laerdal, 2024-11-27:
> @ksidirop-laerdal Have you found any solution to this issue?

Still waiting on more info on this too. Currently it doesn't bother me too much but it's somewhat concerning in general.