nordicsemi/Android-nRF-Connect-Device-Manager #201: Question: In the context of file-uploading and file-downloading what is the enum type of HasReturnCode.GroupReturnCode.group and .rc?

ksidirop-laerdal, 2024-10-25:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/main/mcumgr-core/src/main/java/io/runtime/mcumgr/response/HasReturnCode.java

In our Laerdal.McuMgr bindings-library we need to perform exception-mapping and error-mapping between the Java world and the C# world whenever an error occurs during file-downloading or file-uploading (for example "file not found", "corrupted", etc).

Is it correct to say that in the context of file-uploading an file-downloading any McuMgrErrorException adheres to the following statements?

exception.getGroupCode().rc (int field)          ->  always adheres to 'enum ReturnCode'
exception.getGroupCode().group (int field)   ->  always adheres to 'enum McuMgrErrorCode'

( I just want to be sure that there isn't some corner case where these integer fields are adhering to some different enum because if this happens our error-mapping and exception-mapping will end up mapping and reporting the wrong kind of error over to the C# world misleading the host-application! )

philips77, 2024-10-27:
The `group` is the ID of the manager (group), so that's one of these:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/389c60095a11e685ec3023fc33eed205866f87e2/mcumgr-core/src/main/java/io/runtime/mcumgr/McuManager.java#L60-L76

The `rc` is a the `[TheManagerWithGivenGroupId].ReturnCode`. So, if `exception.getGroupCode().rc == 8` (`FSManager`), than `rc` adheres to 'enum FsManager.ReturnCode', etc.

philips77, 2024-10-27:
If the device doesn't support SMP v. 2, the `exception.getGroupCode()` returns `null`, and you need to use `exception.getCode()`, which adverse to https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/389c60095a11e685ec3023fc33eed205866f87e2/mcumgr-core/src/main/java/io/runtime/mcumgr/McuMgrErrorCode.java#L24

philips77, 2024-10-28:
Hi,
Basic Group is called Zephyr Group, as far as I remember. These are Zephyr specific commands, or actually just one, the erase app settings.

The other group you mentioned aren't supported on Zephyr, thus weren't migrated to SMP V2, so they don't support group errors.

ksidirop-laerdal, 2024-10-28:
> Basic Group is called Zephyr Group, as far as I remember. These are Zephyr specific commands, or actually just one, the erase app settings.

What is the group-id of the basic-group ("zephyr-group"). Can I find it somewhere?

philips77, 2025-08-25:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/8f78431a8b139056080893caa23f909fdb352770/mcumgr-core/src/main/java/io/runtime/mcumgr/McuManager.java#L71-L72

Useful links: 
1. https://github.com/nrfconnect/sdk-zephyr/blob/main/include/zephyr/mgmt/mcumgr/grp/zephyr/zephyr_basic.h
2. https://github.com/nrfconnect/sdk-zephyr/blob/476ad665287a8b92e90ba1ec9a1ebd3241018b7a/include/zephyr/mgmt/mcumgr/mgmt/mgmt_defines.h#L90