nordicsemi/Android-nRF-Connect-Device-Manager #219: Image Erase returns "No Free Slots (9)"

MB-Conti, 2025-01-24:
Hi,
I am experiencing problems with the nrf Device Manager App (v2.4.0). After performing a DFU I want to erase the old image in the secondary slot using the erase button.

Upon selecting "Erase" I receive the feedback "No Free Slots (9)". In the past I have not experienced this.

I am not to firm with app programming, however, in this repo I found that the erase functionality was worked on in the two past releases.

Any chance that this behaviour is a bug?

andybondchiaro, 2025-02-03:
I have seen this as well, when I tried the image erase on an iPhone it was successful.

philips77, 2025-02-04:
My guess is, that the secondary slot, which you want to erase, has _pending_ flag set to true, that is Test or Confirm command was sent, but the device wasn't reset. Check it on Image Control pane in Image tab / Advanced view.

Or, I made a bug in the recent version of the library in #216. Could you paste logs from LogCat or nRF Logger app?

MB-Conti, 2025-02-07:
Hey, the pending flag are false.

My logs are below:

`Device Manager, 2025-02-05
Device (FC:0C:A5:92:A1:CA)
I	10:34:29.429	Connected to FC:0C:A5:92:A1:CA
I	10:34:31.053	Services discovered
I	10:34:31.129	MTU changed to: 498
I	10:34:31.246	Data written to descr. 00002902-0000-1000-8000-00805f9b34fb
I	10:34:31.263	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:31.302	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 01-00-00-19-00-00-FF-06-BF-68-62-75-66-5F-73-69-7A-65-19-09-AB-69-62-75-66-5F-63-6F-75-6E-74-04-FF
I	10:34:31.315	Wait for value changed complete
A	10:34:31.335	Received Header (Version: 0, Op: 1, Flags: 0, Len: 25, Group: 0, Seq: 255, Command: 6) CBOR {"buf_size":2475,"buf_count":4}
I	10:34:31.359	SMP reassembly supported with buffer size: 2475 bytes and count: 4
I	10:34:31.371	PHY read (TX: LE 1M, RX: LE 1M)
A	10:34:31.383	Sending (9 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 1, Group: 0, Seq: 0, Command: 8) CBOR {}
A	10:34:31.396	Sending (9 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 1, Group: 0, Seq: 1, Command: 6) CBOR {}
I	10:34:31.404	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:31.417	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:31.424	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-06-00-00-00-08-BF-62-72-63-08-FF
W	10:34:31.439	Received Header (Version: 1, Op: 1, Flags: 0, Len: 6, Group: 0, Seq: 0, Command: 8) CBOR {"rc":8}
I	10:34:31.454	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-19-00-00-01-06-BF-68-62-75-66-5F-73-69-7A-65-19-09-AB-69-62-75-66-5F-63-6F-75-6E-74-04-FF
A	10:34:31.463	Received Header (Version: 1, Op: 1, Flags: 0, Len: 25, Group: 0, Seq: 1, Command: 6) CBOR {"buf_size":2475,"buf_count":4}
A	10:34:31.472	Sending (9 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 1, Group: 1, Seq: 2, Command: 0) CBOR {}
I	10:34:31.488	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
A	10:34:31.498	Sending (19 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 11, Group: 0, Seq: 3, Command: 7) CBOR {"format":"sv"}
I	10:34:31.514	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:31.546	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-F6-00-01-02-00-BF-66-69-6D-61-67-65-73-9F-BF-64-73-6C-6F-74-00-67-76-65-72-73-69-6F-6E-66-30-2E-31-38-2E-32-64-68-61-73-68-58-20-8E-BB-3B-49-62-F6-F6-CD-1D-29-3C-83-03-E3-7A-78-46-A2-DB-4F-FA-24-FC-9D-D7-D1-F5-B7-B8-BD-B8-6F-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F5-66-61-63-74-69-76-65-F5-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-BF-64-73-6C-6F-74-01-67-76-65-72-73-69-6F-6E-66-30-2E-31-38-2E-32-64-68-61-73-68-58-20-FE-DA-EB-D1-23-3B-30-3E-B9-A0-FD-C0-FF-0E-5E-6B-8B-58-F1-31-9E-E2-41-AD-CD-68-CF-3D-85-A9-51-1E-68-62-6F-6F-74-61-62-6C-65-F5-67-70-65-6E-64-69-6E-67-F4-69-63-6F-6E-66-69-72-6D-65-64-F4-66-61-63-74-69-76-65-F4-69-70-65-72-6D-61-6E-65-6E-74-F4-FF-FF-6B-73-70-6C-69-74-53-74-61-74-75-73-00-FF
I	10:34:31.558	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-06-00-00-03-07-BF-62-72-63-08-FF
A	10:34:31.569	Received Header (Version: 1, Op: 1, Flags: 0, Len: 246, Group: 1, Seq: 2, Command: 0) CBOR {"images":[{"slot":0,"version":"0.18.2","hash":"jrs7SWL29s0dKTyDA+N6eEai20/6JPyd19H1t7i9uG8=","bootable":true,"pending":false,"confirmed":true,"active":true,"permanent":false},{"slot":1,"version":"0.18.2","hash":"/trr0SM7MD65oP3A/w5ea4tY8TGe4kGtzWjPPYWpUR4=","bootable":true,"pending":false,"confirmed":false,"active":false,"permanent":false}],"splitStatus":0}
W	10:34:31.579	Received Header (Version: 1, Op: 1, Flags: 0, Len: 6, Group: 0, Seq: 3, Command: 7) CBOR {"rc":8}
A	10:34:31.599	Sending (9 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 1, Group: 0, Seq: 4, Command: 8) CBOR {}
I	10:34:31.610	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:31.664	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-06-00-00-04-08-BF-62-72-63-08-FF
W	10:34:31.677	Received Header (Version: 1, Op: 1, Flags: 0, Len: 6, Group: 0, Seq: 4, Command: 8) CBOR {"rc":8}
I	10:34:34.520	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
I	10:34:36.782	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
A	10:34:43.741	Sending (15 bytes) Header (Version: 1, Op: 2, Flags: 0, Len: 7, Group: 1, Seq: 5, Command: 5) CBOR {"slot":0}
I	10:34:43.751	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	10:34:43.846	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 0B-00-00-13-00-01-05-05-BF-63-65-72-72-BF-65-67-72-6F-75-70-01-62-72-63-09-FF-FF
W	10:34:43.857	Received Header (Version: 1, Op: 3, Flags: 0, Len: 19, Group: 1, Seq: 5, Command: 5) CBOR {"err":{"group":1,"rc":9}}
I	10:34:44.296	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
I	10:34:48.932	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
`

Rob-Grady, 2025-03-09:
I have seen this also.  I think it is a bug.  The app is passing the slot number to erase as 0, which is the active slot and that fails.  You should set the slot number to 1 in the code or allow user input for the slot number.

You can work around this by modifying the server code on the device to ignore the passed protocol value. 

```
        uint32_t  ignore_slot;
	struct zcbor_map_decode_key_val image_erase_decode[] = {
		ZCBOR_MAP_DECODE_KEY_DECODER("slot", zcbor_uint32_decode, &ignore_slot),
	};
```

in file img_mgmt.c  line 353.