nordicsemi/Android-nRF-Connect-Device-Manager #226: Image upload returns corrupt (9)

Rob-Grady, 2025-03-09:
I'm running version 2.4.1 and everything works well except for the firmware upload.  When I select 'Upload' and 'Image 0 (main)' I immediately get Corrupt (9) as a response.

I have monitored the protocol packets and everything looks correct. I have also successfully uploaded the firmware using the iPhone version and using the PC app "Au Term". 

What would cause this error code?  The device is returning a valid response in the image read message, but it looks like Android app is rejecting it.

Rob-Grady, 2025-03-10:
I have found the problem is that the 'chunk size' that this app is using is too large for my device.  Luckily the iOS version allows you to set the chunk size.  Is there any way to set the size in the android version or force it to use a smaller chunk size?   How does it determine what chunk size to use?  Is there a set value?

philips77, 2025-03-10:
Hello,
Could you point me to where do you set the "chunk size" in the iOS version of this library, so I could better understand what are you talking about?
Is the "chunk size" the GATT MTU? The size of SMP packet? 

Could you also paste some log contanining the data sent and error recevied? I'm intersted in MTU negotiations and the packet size, in bytes.

Rob-Grady, 2025-03-12:
From what I've found, the large packets are reported as 'Corrupt(9)'; this happens on the very first packet containing the first block of the image file data.  (which I assume is a very large packet.)  I was using the default buffer values: 4 buffers, and buffer size 2475 bytes.  These values are passed to the android app, and I assume are used to set the 'chunk' size.  The chunk size is the size of the data block.  The image file is chopped up into chunks, and 1 chunk is sent is every image upload message.

On the iPhone version of the app, select 'image' then 'advanced' then 'chunks'.  You are given the option to enter the 'chunk size' which is also labeled the 'MTU size'.  This is a very helpful option.

I was able to work around the issue by tricking the app.  When it asks for buffer size, I send 256 instead of the actual size of 2475.  This allows for uploading with the corruption error.

The problem could be on the device side (mcumgr server), but I'm not sure.  The device libraries are very hard to see what is going on.  That's why I am wondering what exactly does 'corrupt' indicate.  

The android app may be corrupting the data before it is sent; perhaps it has trouble handling very large buffers? Or maybe the device server is returning a corrupted response. This seems less likely because I never receive notification that the first chunk has been sent.

philips77, 2025-03-12:
Yes, Android app is usnig the MCU Manager Params command to read the buffer size. This actually happens just on the first connection to a device in "transport" layer:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/b4b13ba19d544a7e52dfe2d27070d13a764a572f/mcumgr-ble/src/main/java/io/runtime/mcumgr/ble/McuMgrBleTransport.java#L691-L723

This then is considered the SMP MCU. The SMP packets are split into ATT MTU (usually 498 or less).
Your device should buffer the whole SMP packet (2475 bytes) before passing to upper layer.

It works with the SMP Server sample from nRF Connect SDK.

Rob-Grady, 2025-03-12:
I found the issue!  Thanks to your comment.  My device did not have transport reassembly enabled (CONFIG_MCUMGR_TRANSPORT_BT_REASSEMBLY).  It was responding before the entire packet was sent.  This was causing the corruption.   Thank you so much for your help.

closed 2025-03-12 by Rob-Grady