nordicsemi/Android-nRF-Connect-Device-Manager #227: Please update readme

mobilekosmos, 2025-03-11:
Those setEstimatedSwapTime don't exist anymore as is. At least with Kotlin you must do it like so now:

`dfuManager.start(images, FirmwareUpgradeManager.Settings.Builder()
                            .setEstimatedSwapTime(30000)
                            .build())`

Also this:

You may also use [ZipPackage](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/main/sample/src/main/java/io/runtime/mcumgr/sample/utils/ZipPackage.java) class from the Sample app, which can unpack the ZIP file generated by west in Zephyr or nRF Connect SDK (see [example](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/713a0e76a3765a2f6a417db65f054848e08d7007/sample/src/main/java/io/runtime/mcumgr/sample/viewmodel/mcumgr/ImageUpgradeViewModel.java#L128-L141)).

That linked example if from a git-tag, completely outdated now.

Also why do I need to manually copy-paste the zip utility class, which also needs gson to work, why cannot the lib just handle this for us?

philips77, 2025-03-13:
You're right! Sorry for that. I'll update Readme asap.

> Also why do I need to manually copy-paste the zip utility class, which also needs gson to work, why cannot the lib just handle this for us?

The main reason was, that initialy the ZIP packet format was just a proposal. Users were free to use their own way of delivering the images. That's also why the DFU manager accepts plain byte arrays, not a ZIP file.

However, now, as the ZIP is automatically generated by _west_, perhaps our libs should, like you say, also have it built-in. But then there is a decision to be made - how should the ZIP be unpacked? There is Kotlin serialization, gson, wire, native java API without any dependencies, but with ugly code, etc. Same applies to iOS.

mobilekosmos, 2025-03-13:
Kotlin of course 😊
I think the iOS lib already accepts passing the zip, at least that's what I heard from the other team that implemented it. For the Android implementation we needed to copy the one Zip utility class you have in the repo, but that also depends on gson and this lib, so it's not ideal.

philips77, 2025-03-13:
> I think the iOS lib already accepts passing the zip

You're right. It's using _ZIPFoundation_ dependency:
https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/blob/main/Source/McuMgrPackage.swift

> Kotlin of course 😊

Exactly, for you its Kotlin, for others wire. I'll think about it. I'm busy with other projects atm.