nordicsemi/Android-nRF-Connect-Device-Manager #228: Transaction timer too short for stats retrieval.

Rob-Grady, 2025-03-14:
Here is what I think is happening:

I send  a stats request.

The app retrieves the stats group list.  In my case the list has 10 groups.

The app then sends a group-data retrieval request for each group in the list.

A connection timer is started.

Group-data is for each group is displayed as it is received, one message at a time.

While processing the 5th group, the connection timer expires, and a timeout message is displayed.  

The data from the remaining groups is lost.

I think just lengthening the timer would solve this issue.  Or better yet, restarting the timer after each group-data block is received.

philips77, 2025-03-17:
Hello, thank you for creating the issue. Could you create a PR with required fixes? This way we can have it merged much faster, as I'm busy with other projects.

Rob-Grady, 2025-03-17:
I would like to help because I love this app, but I am not a java programmer nor a Android developer. I don't have the skills or tools for building and testing this app.

I work on the embedded device side, building the devices this app talks to.  If you need help on that side, let me know and I will gladly pitch in.

philips77, 2025-03-24:
Hello,
> Or better yet, restarting the timer after each group-data block is received.

I see in the code, that each request - for th elist of groups and for each module - has a timeout of 2.5 seconds.

Unless you're talking about somethign else then:
https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/d54d7f3c51b4257dbd5676ecf5ee6f2d536a8313/mcumgr-core/src/main/java/io/runtime/mcumgr/managers/StatsManager.java#L132-L134

Would you like to incretase the timeout for each module, or are you talking about some other thing?

Could you also post logs? You can install nRF Logger app from Play Store, read stats and tap on the icon with mignifying glass in top-right corner, and paste it here.

Rob-Grady, 2025-04-09:
Here's the log:

```
A	17:57:35.272	Sending (9 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 1, Group: 2, Seq: 8, Command: 1) CBOR {}
I	17:57:35.295	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	17:57:35.364	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-46-00-02-08-01-BF-62-72-63-00-69-73-74-61-74-5F-6C-69-73-74-9F-69-73-79-73-5F-73-74-61-74-73-68-74-78-5F-73-74-61-74-73-6B-73-6E-69-66-66-5F-73-74-61-74-73-69-74-61-67-5F-73-74-61-74-73-6A-6D-67-6D-74-5F-73-74-61-74-73-FF-FF
W	17:57:35.392	Received Header (Version: 1, Op: 1, Flags: 0, Len: 70, Group: 2, Seq: 8, Command: 1) CBOR {"rc":0,"stat_list":["sys_stats","tx_stats","sniff_stats","tag_stats","mgmt_stats"]}
A	17:57:35.437	Sending (24 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 16, Group: 2, Seq: 9, Command: 0) CBOR {"name":"sys_stats"}
A	17:57:35.457	Sending (23 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 15, Group: 2, Seq: 10, Command: 0) CBOR {"name":"tx_stats"}
A	17:57:35.474	Sending (26 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 18, Group: 2, Seq: 11, Command: 0) CBOR {"name":"sniff_stats"}
A	17:57:35.488	Sending (24 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 16, Group: 2, Seq: 12, Command: 0) CBOR {"name":"tag_stats"}
A	17:57:35.500	Sending (25 bytes) Header (Version: 1, Op: 0, Flags: 0, Len: 17, Group: 2, Seq: 13, Command: 0) CBOR {"name":"mgmt_stats"}
I	17:57:35.512	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	17:57:35.529	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	17:57:35.544	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-34-00-02-09-00-BF-64-6E-61-6D-65-69-73-79-73-5F-73-74-61-74-73-66-66-69-65-6C-64-73-BF-6A-75-70-74-69-6D-65-5F-6D-69-6E-19-01-D0-6A-62-61-74-74-65-72-79-5F-6D-56-15-FF-FF
I	17:57:35.556	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
A	17:57:35.572	Received Header (Version: 1, Op: 1, Flags: 0, Len: 52, Group: 2, Seq: 9, Command: 0) CBOR {"name":"sys_stats","fields":{"uptime_min":464,"battery_mV":21}}
I	17:57:35.586	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	17:57:35.612	Data written to da2e7828-fbce-4e01-ae9e-261174997c48
I	17:57:35.632	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-5D-00-02-0A-00-BF-64-6E-61-6D-65-68-74-78-5F-73-74-61-74-73-66-66-69-65-6C-64-73-BF-68-74-78-5F-74-6F-74-61-6C-19-0D-CC-67-74-78-5F-6D-67-6D-74-19-02-AF-67-74-78-5F-64-61-74-61-19-0A-FE-66-74-78-5F-63-61-6C-18-1F-6A-74-78-5F-64-72-6F-70-70-65-64-00-69-70-68-79-5F-74-6F-74-61-6C-19-0D-CB-FF-FF
A	17:57:35.643	Received Header (Version: 1, Op: 1, Flags: 0, Len: 93, Group: 2, Seq: 10, Command: 0) CBOR {"name":"tx_stats","fields":{"tx_total":3532,"tx_mgmt":687,"tx_data":2814,"tx_cal":31,"tx_dropped":0,"phy_total":3531}}
I	17:57:35.675	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 420ms)
I	17:57:35.686	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-5E-00-02-0B-00-BF-64-6E-61-6D-65-6B-73-6E-69-66-66-5F-73-74-61-74-73-66-66-69-65-6C-64-73-BF-68-72-78-5F-74-6F-74-61-6C-19-47-00-6B-72-78-5F-66-69-6C-74-65-72-65-64-19-26-F4-6A-72-78-5F-62-61-64-5F-63-72-63-00-6A-72-78-5F-64-72-6F-70-70-65-64-00-6B-72-78-5F-61-63-63-65-70-74-65-64-19-20-0C-FF-FF
A	17:57:35.697	Received Header (Version: 1, Op: 1, Flags: 0, Len: 94, Group: 2, Seq: 11, Command: 0) CBOR {"name":"sniff_stats","fields":{"rx_total":18176,"rx_filtered":9972,"rx_bad_crc":0,"rx_dropped":0,"rx_accepted":8204}}
I	17:57:35.708	Notification received from da2e7828-fbce-4e01-ae9e-261174997c48, value: (0x) 09-00-00-5E-00-02-0C-00-BF-64-6E-61-6D-65-69-74-61-67-5F-73-74-61-74-73-66-66-69-65-6C-64-73-BF-68-72-78-5F-74-6F-74-61-6C-19-1A-3D-6B-72-78-5F-66-69-6C-74-65-72-65-64-00-6C-72-78-5F-64-75-70-6C-69-63-61-74-65-19-14-BE-6A-72-78-5F-64-72-6F-70-70-65-64-00-6B-72-78-5F-61-63-63-65-70-74-65-64-19-05-7F-FF-FF
A	17:57:35.723	Received Header (Version: 1, Op: 1, Flags: 0, Len: 94, Group: 2, Seq: 12, Command: 0) CBOR {"name":"tag_stats","fields":{"rx_total":6717,"rx_filtered":0,"rx_duplicate":5310,"rx_dropped":0,"rx_accepted":1407}}
I	17:57:40.585	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 420ms)

```

Rob-Grady, 2025-04-09:
It looks like all 5 queries are sent at the same time, and their timeout timers are started at the same time.  But by the time the last response is received, that timer has expired due to all the processing time for the previous queries.

philips77, 2025-04-11:
OK, I know where ther issue may be. I'l check why the requests are sent simultanously, before waiting for the response.

PR #256 (2025-08-26): Reading stats sequentially, not in parallel
This PR fixes #228.

Also, stats timeout have been increased to MEDIUM (5 seconds).