nordicsemi/Android-nRF-Connect-Device-Manager #232: Feature request: Expose BluetoothGatt from McuMgrBleTransport

cogito666, 2025-05-07:
Hello there,

First of all, thanks a lot for this library!
While testing OTA updates, I found the library well-structured and effective.

During the integration with our own ```custom BLE library```, I faced an issue:
```McuMgrBleTransport``` does not expose ```BluetoothGatt```.

I assume this was intentional to avoid misuse since ```Queue mechanism``` is not implemented on Android OS level like iOS.

But, I suggest exposing ```BluetoothGatt```(or at least provide a public getter interface) for the following reason.

1. Gatt is a high level controller over the physical (= real) BLE connection. And this library also seems to acknowledge it, because ```McuMgrBleTransport``` instance should be passed as an argument to ```FirmwareUpgradeManager``` constructor. And it allows developers to use their preferred BLE libraries. => No issue for starting BLE DFU without having to disconnect/scan/connect to create ```McuMgrBleTransport``` instance.
2. However during the dfu process, I checked interesting log.  When ```DFU state``` changes  ```from RESET to CONFIRM```, this library automatically re-connects to the dfu-target device. (I am not sure but maybe because ```device RESET```?) And this re-connection step must be an link-layer (=physical layer) operation. 
3. After the DFU process completed (= confirmed), developers may want to control the firmware-updated device using their BLE library. And for this, it would be better to provide an interface to getting the ```GATT from McuMgrBleTransport```.
>> It would be good for the final App user experience! Not just for dev. 

Thanks again.

philips77, 2025-05-07:
Hi,
Thanks for the nice words.

1. It is possible to extend `McuMgrBleTransport` and keep the `BluetoothGatt` there. You may add a getter there as well.
2. Did you try `CONFIRM_ONLY`? In this mode the library reports success without reconnection. It sends RESET command, awaits disonnection and reports success. It is up to the app to wait for the swap to complete and continue with other stuff. There is no reconnection untill you want to send another message afterwards. The `McuMgrBleTransport` connects automatically when user wants to send a command.
3. See 1 and 2.

Have a look at https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/blob/main/sample/src/main/java/io/runtime/mcumgr/sample/observable/ObservableMcuMgrBleTransport.java where the sample app extends the functionality of the transport object.

cogito666, 2025-05-07:
Appreciate @philips77 for the fast and kind answer.
Since I got the answer, let me close this issue.