nordicsemi/Android-nRF-Connect-Device-Manager #242: FirmwareUpgradeCallback onUpgradeFailed called even if firmware upgrade is successful

unveloper, 2025-08-04:
Hello
I'm implementing the Nordic firmware upgrade using FirmwareUpgradeManager

```kotlin
private fun performNordicDfu(transport: CustomMcuMgrBleTransport, images: ImageSet) {
        transport.requestConnPriority(ConnectionPriorityRequest.CONNECTION_PRIORITY_HIGH)

        dfuManager = FirmwareUpgradeManager(transport, dfuCallback)
        dfuManager.apply {
            setMode(FirmwareUpgradeManager.Mode.TEST_AND_CONFIRM)
            setCallbackOnUiThread(false)
        }

        try {
            val advancedSettings = FirmwareUpgradeManager.Settings.Builder()
                // 10 secondi per nRF52840 come nella sample app.
                .setEstimatedSwapTime(10_000)
                // Pipelining per migliorare la velocità.
                .setWindowCapacity(3)
                // Allineamento memoria per Nordic devices.
                .setMemoryAlignment(4)
                // Non cancellare le impostazioni app.
                .setEraseAppSettings(false)
                .build()
            dfuManager.start(images, advancedSettings)
        } catch (e: Exception) {
            Timber.e(e, "Error during Nordic DFU start")
            _data.update { FirmwareUpdateStatus.Error }
        }
    }
```

The upgrade proceeds correctly but at the end I always get the callback onUpgradeFailed invoked

```
Firmware upgrade failed, state: RESET
io.runtime.mcumgr.ble.exception.McuMgrDisconnectedException
	at io.runtime.mcumgr.ble.McuMgrBleTransport.onServicesInvalidated(McuMgrBleTransport.java:756)
	at no.nordicsemi.android.ble.BleManagerHandler.notifyDeviceDisconnected(BleManagerHandler.java:2034)
	at no.nordicsemi.android.ble.BleManagerHandler$4.onConnectionStateChange(BleManagerHandler.java:2312)
	at android.bluetooth.BluetoothGatt$GattCallback.lambda$onClientConnectionState$3(BluetoothGatt.java:426)
	at android.bluetooth.BluetoothGatt$GattCallback.$r8$lambda$g0wdVhK6mBSTzAIOY21zUwHI--c(Unknown Source:0)
	at android.bluetooth.BluetoothGatt$GattCallback$$ExternalSyntheticLambda10.run(D8$$SyntheticClass:0)
	at android.bluetooth.BluetoothUtils.lambda$executeFromBinder$0(BluetoothUtils.java:343)
	at android.bluetooth.BluetoothUtils$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
	at android.os.Handler.handleCallback(Handler.java:1041)
	at android.os.Handler.dispatchMessage(Handler.java:103)
	at android.os.Looper.dispatchMessage(Looper.java:315)
	at android.os.Looper.loopOnce(Looper.java:251)
	at android.os.Looper.loop(Looper.java:349)
	at android.os.HandlerThread.run(HandlerThread.java:100)
```

If I try with your official app I got the same result and if I check the version it was updated correctly.
I tried to increment the setEstimatedSwapTime but nothing changed.

Version used 2.6.0

philips77, 2025-08-14:
I managed to replicate the issue, I believe. 
It happens when the device disconnects immediately after receiving Reset command, before it's able to reply. 
I'll prepare a PR with a fix in a moment.

closed 2025-08-14 by philips77

unveloper, 2025-08-25:
I tried different estimatedSwapTime, even 240_000 millisecs but the error always happens after 30 secs with TEST_AND_CONFIRM. Is that the expected behaviour?

With CONFIRM_ONLY the error is not thrown

philips77, 2025-08-26:
Does your device start advertsing with the same address after the DFU?
In `TEST_AND_CONFIRM` the lib is expecting to reconnect to the same MAC address to a device running the firmware under test. When connected, it sends CONFIRM to finalize it.

unveloper, 2025-08-26:
We did more tests, the swap time on the device it's likely 35 secs.
And the advertising starts correctly after the swap. It seems that the setEstimatedSwapTime is not working at all since the timeout happens always after 30 secs not matter the value in the setting.

unveloper, 2025-08-27:
Seems that `remainingTime` is always negative so the postDelayed is never executed.
You can also see that the line is not present in the logs already posted https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager/issues/242#issuecomment-3219309626

[image]

philips77, 2025-08-27:
Btw, have a look at #258. Hopefully it won't be a terrible problem for you and others.

PR #246 (2025-08-14): Bugfix: Successful DFU reported as error and success in case of immediate reboot
This PR fixes #242.
PR #262 (2025-08-27): Bugfix: Delaying reconnection in case of hard reset
This PR fixes #242 for the second time.